Ysurac/openmptcprouter #291: How to only forward specific ports to router?

nileio, 2019-01-04:
thanks for the latest updates and changes, it gets better by the day. i been using the solution for a while!!

## Expected Behavior
Configuration to specify one or more ports to forward from the VPS to OpenMPTCProuter.

## Actual Behavior
I can only forward ALL ports, not selective. I only like to forward specific ports for example 80 & 8080.

## Steps to Reproduce the Problem

wonder if there is a solution for this ?

## Specifications

  - OpenMPTCProuter version: v0.44
  - OpenMPTCProuter VPS version: VPS 0.83
 - OpenMPTCProuter platform: RPI2

Ysurac, 2019-01-04:
Shorewall is used on the VPS, you can modify it in ```/etc/shorewall``` for IPv4 and ```/etc/shorewall6``` for IPv6.

afriza, 2019-01-04:
To be precise, edit `/etc/shorewall/rules`

closed 2019-01-06 by Ysurac