Ysurac/openmptcprouter #292: upgrade problem

renaudadu, 2019-01-06:
hello,
i have a upgrade my systeme router  ( 0.44.5 -> 0.45.1) and vps ( 0.83 -> 084)
now i have a message :
Can't get public IP address from ShadowSocks
in the vps log 
i have a message :
ss-server[386]:  2019-01-06 12:33:08 ERROR: block all requests from (IPwan1defaut)

where can i search thank

## Specifications

  - OpenMPTCProuter version: (0.45.1)
  - OpenMPTCProuter VPS version: (0.84)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-01-06:
The VPS was rebooted ?

Ysurac, 2019-01-06:
I updated OBFS in VPS script, may be better now ;)

renaudadu, 2019-01-06:
i have upgrade server ( 0.85) with the router 0.45.1
OBFS active  not good 👎 
OBFS not active ok but no DNS ( 1.1.1.1 ) 

i notice this on the router (0.45.1):
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.239533] mptcp_close: Close of meta_sk with tok 0x6aa3589d
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.246790] mptcp_close: Close of meta_sk with tok 0x37172195
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.246798] mptcp_close: Close of meta_sk with tok 0x19ecc266
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.254389] mptcp_del_sock: Removing subsock tok 0x6aa3589d pi:2 state 7 is_meta? 0
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.261344] mptcp_del_sock: Removing subsock tok 0x37172195 pi:1 state 7 is_meta? 0
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.271359] mptcp_del_sock: Removing subsock tok 0x6aa3589d pi:1 state 7 is_meta? 0
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.281327] mptcp_sock_destruct destroying meta-sk token 0x37172195
Sun Jan  6 15:48:51 2019 kern.err kernel: [  139.291293] mptcp_sock_destruct destroying meta-sk token 0x6aa3589d

i downgrade the router to 0.44.5 and server (0.85)
OBFS active : not good 👎 
OBFS not actve OK 👍

Ysurac, 2019-01-06:
I found the issue with OBFS, this will be fixed in next release.

closed 2019-01-06 by Ysurac