Ysurac/openmptcprouter #293: Port forwarding and OBFS on last version

ThibautBascoulergue, 2019-01-06:
## Expected Behavior

Since the last version, 45.1 on the router side and 0.87 for the vps. I feel that the port forwarding part does not work. The rules are there and the box "open all port from vps to openmptcprouter" but I do not access from outside.

Another problem but has nothing to do with the port forwarding part, when I check the obfs box, I can not connect the vps tunnel.

## Actual Behavior
For port fortwarding 👍 
[image]

with OBFS unchecked 👍 
[image]

with OBFS checked 👍 
[image]

## Specifications

  - OpenMPTCProuter version: (45.1)
  - OpenMPTCProuter VPS version: (0.87)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-01-07:
Port forwarding will be fixed in next release that is compiling now.
I can't reproduce the issue with obfs for now. Can you check on the VPS, in ```/etc/shadowsocks-libev/config.json``` that there is a *plugin* line with obfs ? If not, how do you activate OBFS ?

YoppY14, 2019-01-07:
I think I have the same issue with OpenMPTCProuter version > 0.44. 
The port forwarding seems does not work.

router : openmptcprouter v0.45.3 r0+8982-a54129d8aa / LuCI Master (git-19.007.24858-12a93ac)
VPS : 0.87

Ysurac, 2019-01-07:
Port forwarding should be fixed in 0.45.3