Ysurac/openmptcprouter #297: Since last router update : Can't contact server admin script

trassiorf, 2019-01-07:
[image]
Since update of router I have always this message "Can't contact server admin script" 

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.45.3
  - OpenMPTCProuter VPS version: (Last is not a version) 0.82
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3

Ysurac, 2019-01-07:
You should update VPS too.

trassiorf, 2019-01-07:
No Why it's necessary

trassiorf, 2019-01-07:
Excuse me the last version of my vps is 0.83 and not 0.82

trassiorf, 2019-01-07:
OpenMPTCProuter version: (Last is not a version) 0.45.3
    OpenMPTCProuter VPS version: (Last is not a version) 0.83
    OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3 B+

Ysurac, 2019-01-07:
Yes, update is needed to have all fully working.

trassiorf, 2019-01-07:
Same thing after update restart of VPS & ROUTER

Curiously I always have to restart the service "Glorytun" by hand

Is it normal for this option (ss_tunnel.dns) to be disabled by default?
[image]

trassiorf, 2019-01-07:
root@scw-1839b3:~# ps aux | grep omr
root       522  0.0  0.3  19840  3576 ?        Ss   11:53   0:13 /bin/bash /usr/local/bin/omr-service
root       855  0.1  3.1  91016 31948 ?        Ss   11:53   0:25 python3 /usr/local/bin/omr-admin.py
it's ok for running service
Not error in /var/log/daemon.log related to omr-admin.py

Ysurac, 2019-01-07:
You checked that correct key is used ?
I will release a new version with a more detailed message :)

trassiorf, 2019-01-07:
Since updating the vps I can not log into ssh on vps anymore from the router.
I have to connect from another adsl box
The keys are identical to the first configuration.
The data was saved and restored identically.
I will test on another vps
I will choose a Kimsufi instead of the Scaleway

trassiorf, 2019-01-07:
Question : 
Différence entre Glorytun et MLVPN ?
Sont-ils indépendants au niveau fonctionnement ?
Le focntionnement est ok sur Kimsufi 
[image]
Mais maintenant j'ai le message "VPN TUNNEL DOWN"

Ysurac, 2019-01-07:
Par défaut Glorytun TCP est utilisé. Il profite de MPTCP pour l'agrégation.
Glorytun UDP permet l'agrégation de lui même en utilisant UDP.
MLVPN utilise aussi UDP et fait également l'agrégation de lui même.
Chaque VPN est indépendant.

trassiorf, 2019-01-07:
Bon étrange fonctionnement ok après seconde réécriture d'image routeur 
J'ai pas tout compris mais ça fonctionne :)
[image]

Ysurac, 2019-01-07:
Il y a un script sur le VPS qui vérifie si le VPN fonctionne, si il fonctionne pas depuis un certain temps il est relancé côté serveur. Donc parfois les soucis se règlent tout seul :)

trassiorf, 2019-01-07:
Le VPS est-il uniquement atteignable en ssh via l'ip du vpn ??  10.255.255.1 par ce qu'en utilisant cette $ip ça fonctionne bien.
En ce qui concerne les 2 différent VPN (Glorytun & MLVPN) quels avantages à tirer de l'un vis à vis de l'autre ?

Ysurac, 2019-01-07:
Non, il est accessible via son IP directement.
Un ```ip route``` sur OpenMPTCProuter doit permettre de voir des routes directes vers le VPS. Son IP est également "bypassée" dans ```ipset list```. Si tout est bon, ça peut être le serveur qui bloque via un fail2ban ou autre...

Sinon les VPNs c'est une question de choix, certains peuvent fonctionner mieux que d'autres dans certains cas.

trassiorf, 2019-01-07:
Une  question comment puis-je forcer simplement la résolution vers les dns opendns ?

Ysurac, 2019-01-07:
Dans réseau->DHCP puis *transmissions DNS*.
Mais c'est souvent une mauvaise idée d'utiliser OpenDNS ;)

trassiorf, 2019-01-07:
Pourquoi c'est une mauvaise idée pour opendns ?
Les résolvers utilisés en temps sont par défaut ceux du serveur VPS ? ou ceux du fournisseurs d'accès ?

Ysurac, 2019-01-07:
Opendns est un DNS menteur qui vend les données liées aux requêtes.
Par défaut les DNS racines sont utilisés.

trassiorf, 2019-01-07:
Je cherche à protéger mes enfants et je ne voyais qu' Opendns pour les protéger
Donc DNS racines c'est ceux du fournisseur d’accès je pense ? donc par défaut les requêtes sont tracées

closed 2019-01-07 by trassiorf

Ysurac, 2019-01-07:
Il y a peut être un paquet openwrt qui fait ça mieux... Je n'ai jamais eu cette problématique.