Ysurac/openmptcprouter #298: Update to 0.45.2 fail

mycanaletto, 2019-01-07:
Hello,
The update of the router in 0.44.5 in 0.45.1 or 2 is not going well. The router reboots but it is impossible to connect to the interface or the internet. Restoration from ok from Veeam 
[image]
in 44.5.

## Steps to Reproduce the Problem

  1.Update from the interface (write firmware)
  2.Reboot OK
  3.Hight CPU

## Specifications

  - OpenMPTCProuter version: 0.44.5
  - OpenMPTCProuter VPS version: 0.87
  - OpenMPTCProuter platform: ESXi

Ysurac, 2019-01-07:
You decompressed the image before update ? Else this fail from unknow reason...

mycanaletto, 2019-01-07:
No, as usual I directly upload the file: openmptcprouter-v0.45.3-r08982-a54129d8aa-x86-64-combined-squashfs.img.gz

Ysurac, 2019-01-07:
I'm not able to reproduce the issue on virtualbox.
Do you have something on ESXi console ?

Ysurac, 2019-01-07:
Update from <0.45 seems to fail. I try to find why.

mycanaletto, 2019-01-07:
I have nothing on the console after the update if it is a hight cpu. I'm not sure I have SSH access either. What can I do on the console to help you?

Ysurac, 2019-01-07:
I hope this will be better in next version that is compiling...

mycanaletto, 2019-01-08:
0.45.6 is OK ?

Ysurac, 2019-01-09:
If you want to update via the interface no. You can save the config and import it again in new version.

closed 2019-03-13 by Ysurac