Ysurac/openmptcprouter #299: vpn IP problem after upgrade

renaudadu, 2019-01-07:
hello,
i have a new problem with vpn ( internet -> LAN ipsec IKEV2)
i think there is problem udp 
in the router log i see 
Mon Jan  7 20:37:22 2019 daemon.info /usr/bin/ss-redir[3647]: UDP relay enabled
Mon Jan  7 20:37:22 2019 daemon.err /usr/bin/ss-redir[3647]: [udp] getaddrinfo: System error

i think there is a problem with the ss-redir for UDP

have you an idee

thank you for your help and your project.

 
## Specifications

  - OpenMPTCProuter version: (0.45.3)
  - OpenMPTCProuter VPS version: (0.87)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-01-07:
By default shadowsocks UDP is a failover if the VPN is down. So should'nt make a problem here.

renaudadu, 2019-01-07:
i  notice that with the old version 0.44.5 router and server 0.83
if ss-redir for UDP is unset VPN is ok
if ss-redir for UDP is hi -tcp and udp vpn is KO
bout now vpn is alway ko