Ysurac/openmptcprouter #300: Gateway down (LTE Gateway doesn't respond to ping)

linuzel, 2019-01-08:
Hi,

I am using OMR on a pcengines apu2 with 2 main WANs (eth1 and eth2) and an LTE backup (3g-lte) via the internal mPCIe/USB port.

The LTE link is up and working as omr-test-speed 3g-lte works but because the gateway does not respond to ping it does not seems to be working via OMR/MPTCP.

Is there a way to change the address used to test if the link is up ? For example try to contact the VPS directly.

I beleave it might be related to the closed issue #259 but I can't test using static IP as recommended there as this is a pppoe link.

## Expected Behavior
Set the LTE interface to monitor another address ro detect if it's up

## Actual Behavior
The interface is not working due to "Gateway DOWN"

## Steps to Reproduce the Problem
Use an LTE carrier that does not allow the gateway to respond to ping

## Specifications

  - OpenMPTCProuter version: 0.45.5
  - OpenMPTCProuter VPS version: 0.87
  - OpenMPTCProuter platform: 86_64_generic UEFI on APU2c4

[image]

Ysurac, 2019-01-08:
First time that a gateway really doesn't respond to ping... This will need to change some code.

linuzel, 2019-01-08:
Yeah I don't understand why, sadly I have no control over it.
Let me know if you need anything.

oOSphaXOo, 2019-01-08:
> First time that a gateway really doesn't respond to ping... This will need to change some code.
Hi, I don't think that an isolate issue because I have the same problem when I use a share connection with my GSM (I try with 2 different Android Phone). I can connect it (USB or Wifi) but OpenMPTCProuter don't ping the gateway. When  I try share connection with a Windows / Linux PC I can ping the gateway (192.168.42.129 or 192.168.43.129 hardcoded in Android)...
Let me know too if you need some log to help you

linuzel, 2019-02-11:
Hello,
I closed this issue by mistake, it is still present in the current release.

Ysurac, 2019-02-11:
The warning is still visible on status page (gateway really doesn't answer) but this should make any problem.
What problem do you have ?

linuzel, 2019-03-05:
The problem i have is that it omr-tracker often detect the link as down when it is not, making the switch to the backup fail.
I am not running OMR at the moment so i can't try again and collect logs, maybe it is solved and i made a mistake somewhere, i think you can close this issue as it does no seems to affect anybody but me and even i am not sure it is still present.

closed 2019-03-05 by linuzel