Ysurac/openmptcprouter #304: Raspberry Pi 3B generated problems, VPN TUNNEL DOWN, and OMR 6in4 NOip NO gateway

suyuan168, 2019-01-09:
## Expected Behavior

   In English please.

## Actual Behavior
I am using Raspberry Pi 3B, broadband overlay appears, VPN TUNNEL DOWN

## Steps to Reproduce the Problem

   1.VPN TUNNEL DOWN
   2.OMR 6in4 NOip NO gateway

## Specifications

   - OpenMPTCProuter version: (Last is not a version)NEW
   - OpenMPTCProuter VPS version: (Last is not a version)NEW
   - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) PRI3B

suyuan168, 2019-01-09:
[image]
This is the status picture

suyuan168, 2019-01-09:
[image]
？

suyuan168, 2019-01-09:
Glorytun  Glorytun can't run, always click start to stop

Ysurac, 2019-01-09:
omr6in4 must have multipath disabled.
For the VPN, you should update and check the system log.

suyuan168, 2019-01-10:
Download speed is particularly slow, isn't it broadband overlay? Upload speed looks normal
[image]

suyuan168, 2019-01-10:
My single network card is over 20M.,Why is there so little after routing?

trassiorf, 2019-01-10:
[image]
Same thing for me the download BP is very low.
Generally I have almost a BP at 60Mbps out of aggregation

Specifications

   - OpenMPTCProuter version: (Last is not a version) 0.45.6
   - OpenMPTCProuter VPS version: (Last is not a version) 0.87
   - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3 B+

suyuan168, 2019-01-10:
Are you also RPI 3B? My remote server VPS port is 100M delayed by 30ms, and the local single 4G speed is 20M. After using two cards to aggregate, the speed is slower? Why is that? ?

trassiorf, 2019-01-10:
Yes RPI 3B "+" local 4 G (WAN2) speed is over 60M (Day) over 80M (Night)
 I use only one network card this is default of RPI3 B+
Latency average on Scaleway VPS is on 36ms
WAN1 is on the switch with local $IP 192.168.11.1 BP below than 4.5Mbps
 Since version 0.45.6 WAN1 never go up

suyuan168, 2019-01-10:
What is going on here? How to do? There seems to be no better solution to the problem now?

trassiorf, 2019-01-10:
suyuan168 : "After using two cards to aggregate, the speed is slower? Why is that? ?" 
Why use 2 cards for aggregate ?
See only one is needed
https://docs.ovh.com/fr/overthebox/installation-manuelle-otb-intel/
[image]

suyuan168, 2019-01-10:
I am now more than 4G Huawei E8372

trassiorf, 2019-01-10:
Ok no switch
I use old Huawei B593u12 so no problem for me

suyuan168, 2019-01-10:
Thu Jan 10 17:32:14 2019 daemon.err glorytun[16955]: tun write: I/O error
[image]

suyuan168, 2019-01-10:
there is a problem. After a few hours, I prompted daemon.err glorytun[16955]: tun write: I/O error

suyuan168, 2019-01-10:
My Glorytun always stops unexpectedly. Need to start manually. Is this Glorytun UDP or TCP? How to set it up? Thu Jan 10 21:47:03 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Jan 10 21:47:03 2019 daemon.err glorytun[20922]: couldn't create tun device
Thu Jan 10 21:47:13 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Jan 10 21:47:13 2019 daemon.err glorytun[21333]: couldn't create tun device
Thu Jan 10 21:47:15 2019 daemon.err /usr/bin/ss-local[17691]: ​​getpeername: Socket not connected
Thu Jan 10 21:47:43 2019 daemon.err glorytun[22004]: couldn't create tun device
Thu Jan 10 21:48:12 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Jan 10 21:48:12 2019 daemon.err glorytun[22610]: couldn't create tun device
Thu Jan 10 21:48:14 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Jan 10 21:48:14 2019 daemon.err glorytun[22699]: couldn't create tun device
Thu Jan 10 21:48:15 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Jan 10 21:48:15 2019 daemon.err glorytun[22809]: couldn't create tun device
Thu Jan 10 21:48:17 2019 daemon.err /usr/bin/ss-local[17691]: ​​getpeername: Socket not connected

error. And MLVPN is also in the DOWN state. How is it unstable? How can I solve it?

trassiorf, 2019-01-10:
Have you test ? : The permutation of Glorytun protocol tcp to udp then restart service

suyuan168, 2019-01-10:
Tested or stopped abnormally. I think it has something to do with this MPTCP---settings---Congestion Control. But I don't know how to set it up.

Ysurac, 2019-01-11:
I will try to reproduce this issue.

suyuan168, 2019-01-12:
> I will try to reproduce this issue.

Thank you

closed 2019-03-13 by Ysurac