Ysurac/openmptcprouter #305: Unstable connection & port forward

quanticware, 2019-01-09:
Hello,

I installed OMR 45.3 yesterday and I tried to made a port forward, but nothing work.
This is what I configured:
[image]
[image]

Redirects all ports from server to OpenMPTCProuter is checked.

Behind 192.168.100.196, I have a Asus router configured to port forward  to haproxy container.

Today I tried to updated to  0.45.6 after upgrade, no working port forward, and very very unstable connection:
[image]
Green -> Orange -> Red -> Green -> Orange -> Red etc.

WAN IP from OpenMPTCPRouter Server, gone and back, I cannot navigate on website more than 30 seconds to get a timeout and unable to go on speedtest, at each test, I have download error.

What I can do? thanks

## Specifications

  - OpenMPTCProuter version: 0.45.6
  - OpenMPTCProuter VPS version: 0.87
  - OpenMPTCProuter platform: RPI3

quanticware, 2019-01-10:
After downgrade to 0.45.3, port forward works.

But always very unstable.

VPS is host by firstheberg.

Ysurac, 2019-01-10:
This issue Will be solved asap : tomorrow or this week end

quanticware, 2019-01-11:
Thanks Grand Canard Blanc!

quanticware, 2019-01-12:
I just upgrade to 0.45.9, connection is stable! Fine! Thanks.

Just one thing, I tried severals reboots of Rpi3 and at each time, I need to enable ipv6 in Settings Wizard otherwise no ipv6 access alert on status monitor.

Thanks again!

closed 2019-01-20 by Ysurac