Ysurac/openmptcprouter #308: shadowsocks is not running

Yanaem, 2019-01-10:
## Expected Behavior

shadowsocks runs

## Actual Behavior
[image]

shadowsocks is not running

## Steps to Reproduce the Problem

Could you tell me how to activate it?

## Specifications

  - OpenMPTCProuter version: (V0.45)
  - OpenMPTCProuter VPS version: (V0.87)
  - OpenMPTCProuter platform: (RPI3 B+)

Yanaem, 2019-01-10:
When i try to merge two internet (4g+lan) it's say ok but the speedtest is less than 1 alone...
[image]

Ysurac, 2019-01-11:
I made a mistake in config file...
You can go in Services->shadowsocks-libev, edit all ss-redir entry and save. This will fix the config file.
Else you can wait for the next release.