Ysurac/openmptcprouter #315: Catastrophic rate since last update

Roken120, 2019-01-13:
## Expected Behavior

My download rate is very slow since last update

## Actual Behavior

## Steps to Reproduce the Problem

  1.Update vps and router
  2.
  3.

## Specifications

  - OpenMPTCProuter version: 0.45.9
  - OpenMPTCProuter VPS version: 0.88
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) x86_64

Ysurac, 2019-01-13:
Can you give me speed and type of each wan and the result of average speed with ```omr-test-speed``` via SSH on OpenMPTCProuter ?

Roken120, 2019-01-13:
After vps reboot hi work normaly maybe the port forwarding not work....

Roken120, 2019-01-13:
6 10.0G    6  688M    0     0  22.6M      0  0:07:31  0:00:30  0:07:01 26.1M

Roken120, 2019-01-13:
I have update to 0.45.10 maybe hi not work

Roken120, 2019-01-13:
After many reboot hi rework

suyuan168, 2019-01-15:
> 在多次重新启动之后，重新工作

How do you restore the download speed? After I updated it, the download speed was very slow. The upload speed is normal.

Roken120, 2019-01-15:
No I'm doing the new thing to work better after a new reboot

Roken120, 2019-01-15:
After reboot he not wok with 0.45.10

Roken120, 2019-01-15:
I have same error with 0.45.11 for router and 0.89 for vps

Ysurac, 2019-01-15:
Do you have IPv6 enabled ? If yes, try to disable it for now.
Check also the congestion control used in Network->MPTCP interface.

Roken120, 2019-01-15:
congestion control : olia

Ysurac, 2019-01-15:
No, but you can give me the result of ```uci show``` on pastebin.com (don't forget to remove your server IP at least).

Roken120, 2019-01-15:
https://pastebin.com/JGKXFs1p

Roken120, 2019-01-15:
Sorry for late

Ysurac, 2019-01-16:
* wan1 and wan2 are on the same network
* multipath is disabled on usb interface (no idea if it's what you want)

Roken120, 2019-01-16:
for the USB interface it's normal, I had a correct flow with this config even before I change anything. I just updated

Roken120, 2019-01-17:
I still have the same problem with the new updated

Ysurac, 2019-01-17:
Do you fixed your configuration and put wan1 and wan2 on different networks ?

suyuan168, 2019-01-18:
I am a USB 4G wireless network card, such as the 4G service of three different operators of Huawei E8372. I did not use wan to plug in the cable. My main equipment is for outdoor use. I don't know if it is related to this.

suyuan168, 2019-01-18:
I need multipath is on usb interface, I really need it. Because I am a USB 4G network card, I get the most stable network connection through different carrier services. With maximized broadband.

Roken120, 2019-01-18:
I have fixed the bug with reset all router config, my connection he rework !

closed 2019-02-05 by Ysurac

Roken120, 2019-02-09:
Hello,
I have the same problem with new version : my download dont up limited to 60-70mbps