Ysurac/openmptcprouter #317: VPN down / OMR error

trassiorf, 2019-01-14:
VPN tunnel down since 0.45.10
I can't stat Glorytun manually 

this is logs of system :
Mon Jan 14 09:02:59 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:03:13 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:03:26 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:03:40 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:03:54 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:04:07 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:04:21 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection
Mon Jan 14 09:04:34 2019 daemon.err omr-tracker-ss[12726]: Failed to establish connection

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.45.10
  - OpenMPTCProuter VPS version: (Last is not a version) 0.88
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3B+
[image]

Ysurac, 2019-01-14:
Error ``` omr-tracker-ss[12726]: Failed to establish connection``` will be fixed in next version.
What is the VPN used ? (Glorytun TCP, Glorytun UDP, OpenVPN or MLVPN ?)

trassiorf, 2019-01-14:
Solve my problem new flash without "keep settings" 
[image]

trassiorf, 2019-01-14:
idem on version 0.45.10
[image]

trassiorf, 2019-01-14:
ok on Scaleway VPS also
[image]

closed 2019-01-14 by trassiorf