Ysurac/openmptcprouter #318: Replace Simple-OBFS by Shadowsocks V2-Ray plugin

Ysurac, 2019-01-14:
simple-OBFS is deprecated.
Use https://github.com/shadowsocks/v2ray-plugin instead.
Need to create packages, modify ShadowSocks interface and add support in VPS script.

Vinc32du44, 2019-01-14:
Good news :-D
When will it be aproximately available ?

Ysurac, 2019-01-20:
This will be difficult : I need to get v2ray compiling, and this need golang protobuf and golang go genproto. For now I have a circular dependency... I need to find how to make a working OpenWrt package from source.

Ysurac, 2019-01-26:
For now, binary distribution of v2ray will be used.
So v2ray will be available on supported platform in next release (0.46).

Vinc32du44, 2019-02-05:
v2ray plugin Ok but with this parameter activated i got "Can't get public IP adress from ShadowSocks" error on VPS.
I update both VPS and MPTCP ...
Do you have a suggest ;)
Thanks

Ysurac, 2019-02-05:
VPS is updated and rebooted to latest version ?

Vinc32du44, 2019-02-05:
Yes rebooted but not after activate v2ray Obfuscating option.
I'll try to reboot the VPS ;)

Vinc32du44, 2019-02-05:
After reboot all is GooD :))) but always no IPV6.
All work fine ! Thanks Ycarus

Ysurac, 2019-02-05:
Yes there is a bug in setting page that enable IPv6, this will be fixed in next release tomorrow.

closed 2019-12-20 by Ysurac