Ysurac/openmptcprouter #319: DNS problem

renaudadu, 2019-01-14:
hello,
i have a DNS problem
i use the openMPTCPROUTER DNS service and i have some resolving problem.
for some site it's working and other not.
i try with 1.1.1.1 or 8.8.8.8 dns it's working.
ss-redir for UDP : is unset.

where can i see where is the problem

## Specifications

  - OpenMPTCProuter version: (0.45.10)
  - OpenMPTCProuter VPS version: (0.88)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-01-14:
The VPN is working ?
You need also to have no DNS server defined for each WAN.

renaudadu, 2019-01-14:
vpn is running
and no dns on each wan.

it's possble to forward dns request to google dns for example ?
if it's  possible omr bypass working ?

Ysurac, 2019-01-14:
Yes in *Network->DHCP & DNS* and *DNS forwardings*
But some public DNS are used by OMR-Tracker to test if a WAN is up or down, this can cause some conflict in some cases.

renaudadu, 2019-01-14:
ok thank.
i modify omr-tracker and i forward to my dns.

closed 2019-01-20 by Ysurac