Ysurac/openmptcprouter #323: SDCard System ReadOnly

MrTheSkream, 2019-01-17:
## Actual Behavior

Hello, I have a problem with the file system, when there is a power failure the system is read-only.

## Steps to Reproduce the Problem

  1. Connect your raspberry
  2. Unplugged the power cable
  3. And restart raspberry

## Specifications

  - OpenMPTCProuter version: ALL VERSION
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-01-17:
Power failure can create FS errors. OpenWrt, on what OpenMPTCProuter is based doesn't do a e2fsck on ext4 at boot. I will check if I can add that.

MrTheSkream, 2019-01-18:
Okey thanks

MrTheSkream, 2019-01-18:
Would you have the command to run to avoid read-only ?

afriza, 2019-01-31:
@Ysurac do you know how to manually fix this?

```
root@OpenMPTCProuter:~# e2fsck -p /dev/mmcblk0
/dev/mmcblk0 is in use.
e2fsck: Cannot continue, aborting.
```

closed 2019-07-04 by Ysurac