Ysurac/openmptcprouter #324: VPN problem

yoyosm2000, 2019-01-17:
## Expected Behavior

  In English please.

## Actual Behavior
First problem: unable to connect to Glorytun VPN. I can use MLVPN but not Glorytun (VPN down). I have check the password and encryption and it's good... Don't understand why it's not working

Second problem: In OPMR state, I have this for VPS: Can't contact server Admin Script (No token). I have check on the VPS and omr-admin.pyis running. So I can't use OpenMPTCProuter VPS key but need to use individual passwors for each service.

Thrid problem: I have this in log, every second:
Thu Jan 17 10:36:40 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jan 17 10:36:40 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jan 17 10:36:40 2019 daemon.notice netifd: omr6in4 (14985): Command failed: Unknown error
Thu Jan 17 10:36:40 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jan 17 10:36:40 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jan 17 10:36:40 2019 daemon.notice netifd: omr6in4 (15012): Command failed: Unknown error

## Steps to Reproduce the Problem

  1. OpenMPTCPRouter wizard with 2 Wan connexions. Shadowsocks is OK. IPv6 is disabled.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: v0.45.12 r0+9007-529c95cc15
  - OpenMPTCProuter VPS version: 0.89 on FirstHeberg VPS
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-01-17:
Check Admin VPS key, if it's correct check VPS log about glorytun or anything strange here.

yoyosm2000, 2019-01-17:
Thanks for your answer and your help.

I'm sure of the Admin VPS key, I've copy/paste it from the openmptcprouter_config.txt on the VPS.

And even with the specific Glorytun key, the problem is the same.
Where can I check Glorytun on the VPS ? On the syslog file I don't see anything about it.

And now I have this error on the router:

[image]

I didn't have modify the omr6in4 interface....

Ysurac, 2019-01-17:
The bug with omr6in4 will be fixed in next release.

yoyosm2000, 2019-01-17:
Thank you for your precious help

I don't understand, it seems that tun0 is started by glorytun if I read the log:
[image]

[image]

but if I check the state, it' shows VPN is down...
[image]

and default VPN is glorytun TCP
[image]

Ysurac, 2019-01-17:
VPN is show as UP if the server answer to ping via the VPN. Here this seems to not be the case.

yoyosm2000, 2019-01-17:
I have this in the kernel log, it should be the cause ?

network card is Intel e1000

[image]

yoyosm2000, 2019-01-17:
Sorry for this flood but issue can be closed.

I have solved it using Ubuntu 18.04 on FirstHeberg VPS and all is working perfect. I think there is something missing on the Debian 9 distrib provided by First Heberg.

Thanks again for your help

closed 2019-01-20 by Ysurac