Ysurac/openmptcprouter #326: Significant loss in the aggregation of connections

Tit34200, 2019-01-18:
Hello everyone.
First of all, I'm French and my English is not rich enough in vocabulary I write you this message with google translate, so sorry if there are any imperfections.

I am posting this message to describe a problem of aggregation of three internet connections.

Here are my tests done before you write:
OpenMPTCProuteur v0.45.3 r0 + 8982-a54129d8aa / LuCI Master (git-19.007.24858-12a93ac)
Equipment:
RPI 3B +
Router 1 huawei b715 configured with LTinspector.
Router 2 archer mr200
SFR box ADSL average downstream rate 7 mb / s.
VPS OVH

I have antennas for both operators:
LTE 800
LTE1800
UMTS 900
UMTS 2100

Test speed in wifi with note 9 to 1m routers with the speedtest app on 5Ghz frequencies:
huawei b715 SFR card: 30mb / s going down 55mb / s upward ping 55ms
archer mr200 card bouygues: 25 mb / s down 92 mb / s upward ping 64 ms

max theoretical: 30 + 25 + 7 = 62 mb / s

Test speedtest with the huawei + the archer + SFR BOX: 25 mb / s descending 15.16 mb / s ascending ping 47 ms
During this test the inspector told me that on the huawei it was mounted at 5mb / s, I have no data on the archer, but knowing that the BOX SFR out at my house on average 7 mb / s the archer with Bouygues was at 13 mb / s

max: 25 mb / s

We have a lot of difference, especially that on my tests on the phone, I have OVH as an access provider.
So the only routers connected to the VPS perform more WI FI aggregation, it even has a loss of 59%.
The problem can it come from wiring between them, the RPI or OpenMPTCPr'eur wrongly configured?

What is certain is that the results with openMPTCProuteur are better than when I used a r470 + router in load balancing but I find the loss important.

Well, if you can help me find the problem of this loss it would be great. ;)

Ysurac, 2019-01-18:
I need a capture of Status page, and the result of ```uci show network``` from OpenMPTCProuter by SSH.
You can also check in Network->MPTCP in MPTCP Fullmesh tab if you have all yours IP.
In MPTCP tab congestion control should be set to bbr and path-manager to fullmesh.

Tit34200, 2019-01-18:
Thank you for your help.
Here is the capture and Fullmesh tab, I have all the IP.

[image]
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 6
2, 3, 0, 192.168.10.2 15
3, 4, 0, 192.168.11.2 16
5, 6, 0, 192.168.12.2 28
IPv6, next v6-index: 0

By cons for:
uci show network from OpenMPTCProuter by SSH
and In MPTCP tab congestion control I do not see how to give you this information, sorry.

Tit34200, 2019-01-19:
Here is uci show network

root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fdd2:5535:e536::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='b8:27:eb:cf:e4:e3'
network.lan.metric='2'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.metric='3'
network.wan1.macaddr='72:af:23:16:ed:11'
network.wan1.ipaddr='192.168.10.2'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.10.1'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.type='macvlan'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth0'
network.wan2.metric='4'
network.wan2.macaddr='5a:d6:da:60:54:41'
network.wan2.ipaddr='192.168.11.2'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.11.1'
network.wan3=interface
network.wan3.proto='static'
network.wan3.type='macvlan'
network.wan3.ip4table='wan'
network.wan3.multipath='on'
network.wan3.defaultroute='0'
network.wan3.ifname='wan3'
network.wan3.masterintf='eth0'
network.wan3.metric='5'
network.wan3.macaddr='ea:bf:2a:3c:dc:a7'
network.wan3.ipaddr='192.168.12.2'
network.wan3.netmask='255.255.255.0'
[... 147 lines cut ...]
network.omrvpn_route.table='6'
network.omrvpn_route_default=route
network.omrvpn_route_default.interface='omrvpn'
network.omrvpn_route_default.target='0.0.0.0'
network.omrvpn_route_default.netmask='0.0.0.0'
network.omrvpn_route_default.gateway='10.255.255.1'
network.omrvpn_route_default.table='6'
network.wan2_route_default_metric=route
network.wan2_route_default_metric.interface='wan2'
network.wan2_route_default_metric.target='0.0.0.0'
network.wan2_route_default_metric.netmask='0.0.0.0'
network.wan2_route_default_metric.gateway='192.168.11.1'
network.wan2_route_default_metric.metric='4'
network.wan3_route_default_metric=route
network.wan3_route_default_metric.interface='wan3'
network.wan3_route_default_metric.target='0.0.0.0'
network.wan3_route_default_metric.netmask='0.0.0.0'
network.wan3_route_default_metric.gateway='192.168.12.1'
network.wan3_route_default_metric.metric='5'
network.wan1_route_default_metric=route
network.wan1_route_default_metric.interface='wan1'
network.wan1_route_default_metric.target='0.0.0.0'
network.wan1_route_default_metric.netmask='0.0.0.0'
network.wan1_route_default_metric.gateway='192.168.10.1'
network.wan1_route_default_metric.metric='3'

Tit34200, 2019-01-19:
And Congestion control is of course bbr and path-manager on fullmesh.

[image]

Ysurac, 2019-01-19:
I don't see anything wrong here.
On OpenMPTCProuter, via SSH you can run ```omr-test-speed``` to test speed via the VPS, and ```omr-test-speed wan1```,```omr-test-speed wan2``` and ```omr-test-speed wan3``` to test speed via each WAN (this give speed in KB/Ko or MB/Mo you need to x8 to get Kb/s or Mb/s). You can stop each test after 1 minute using Ctrl+C.

Tit34200, 2019-01-19:
with omr-test-speed I get 2367k or 18.9 mb / s.
For wan 1 I get 970k is 7.76 mb / s
For wan 2 I get 788k is 6.3 mb / s
For wan 3 I get 828k is 6.6 mb / s

So actually the added measures fit well. But at the same time that I tested the wan 3, I tested it on my phone with speedtest and I get 18 mb / s.
It may be normal, but I think it's a lot of gaps.

Ysurac, 2019-01-19:
True, but the problem doesn't seems to be on OpenMPTCProuter side. Speedtest open multiple connections to test speed this can give higher result than real speed on one connection.
But I would check the switch, link,...

Tit34200, 2019-01-19:
Ok, thank you I understand.
Currently my configuration is the following.
RPI connected to the box sfr which is wan1.
wan 1 connected to the huawei which is the wan 2
computer connected to huawei wan2
archer m200 wan3 connected to huawei wan 2
The ethernet cables are new.
If it can help you.
Can the ethernet card on the computer have an influence?

Ysurac, 2019-01-19:
It's something like this:
```

RPI ---- SFR --- HUAWEI -- ARCHER
                   |
              Computer 
````
You should try to connect RPI to HUAWEI too.
The computer doesn't change anything, because tests was done on the RPI.

Tit34200, 2019-01-19:
Yes, my setup is this one.
I tested other montages without change, I do not understand, I arrive with a single connxion wifi on the laptop to have more than the aggregation of 3 connections.
I do not understand anything anymore.

Ysurac, 2019-02-05:
Yes it was possible.

closed 2019-02-05 by Ysurac