Ysurac/openmptcprouter #327: SSH & FTP unusable

quanticware, 2019-01-20:
## Actual Behavior
Hello,
When I use OpenMPTCProuter, I cannot use SSH or FTP protocol, I have lot of timeout in ssh after only few seconds, SCP don't work simply, and FTP disconnect regulary while download.
Need to go back on my Freebox only to work.

## Specifications

  - OpenMPTCProuter version: 0.45.13
  - OpenMPTCProuter VPS version: 0.89
  - OpenMPTCProuter platform: RPI3 x64

Ysurac, 2019-01-20:
I don't have this issue. You can try to disable fastopen in shadowsocks configuration. Or increase Pv4 TCP Keepalive time but 7200 should be enough.
Else you can use SSH ServerAliveInterval setting.

You should also check logs if there is nothing strange.

Ysurac, 2019-01-20:
What is your provider ? Some seems to have problems with TCP Fast Open, and this is a common issue with ShadowSocks (and give this timeout in a few seconds).

quanticware, 2019-01-21:
Hello,
I'm very very sorry, but I don't find settings in interfaces. Where is it? 

I have Free & SFR.

Other thing, after 0.45.15 update, SSH is little bit more stable. 
But, just now, I cannot connect to SSH server, I don't know why, timeout.

closed 2019-03-13 by Ysurac