Ysurac/openmptcprouter #328: Unstable Glorytun VPN !

Vinc32du44, 2019-01-20:
## Actual Behavior

Update VPS and OpenMPTCPRouter to the last version and reboot both but : 

Sun Jan 20 12:28:35 2019 daemon.err /usr/bin/ss-redir[6804]: remote recv: Connection reset by peer
Sun Jan 20 12:28:35 2019 daemon.err /usr/bin/ss-redir[6803]: remote recv: Connection reset by peer
Sun Jan 20 12:29:58 2019 daemon.err /usr/bin/ss-redir[6803]: remote recv: Connection reset by peer
Sun Jan 20 12:31:19 2019 daemon.err glorytun[5679]: read: Connection reset by peer
Sun Jan 20 12:31:19 2019 daemon.info glorytun[5679]: STOPPED tun0
Sun Jan 20 12:31:20 2019 daemon.err glorytun[5679]: 51.15.xxxxx.65001: connected
Sun Jan 20 12:31:20 2019 daemon.info glorytun[5679]: STARTED tun0
Sun Jan 20 12:31:26 2019 daemon.info glorytun[5679]: STOPPED tun0
Sun Jan 20 12:31:27 2019 daemon.err glorytun[5679]: 51.15.xxxxx.65001: connected
Sun Jan 20 12:31:27 2019 daemon.info glorytun[5679]: STARTED tun0
Sun Jan 20 12:31:31 2019 daemon.err /usr/bin/ss-redir[6803]: remote recv: Connection reset by peer
Sun Jan 20 12:32:08 2019 daemon.err glorytun[5679]: read: Connection reset by peer
Sun Jan 20 12:32:08 2019 daemon.info glorytun[5679]: STOPPED tun0
Sun Jan 20 12:32:09 2019 daemon.err glorytun[5679]: 51.15.xxxxx.65001: connected
Sun Jan 20 12:32:10 2019 daemon.info glorytun[5679]: STARTED tun0
Sun Jan 20 12:32:26 2019 daemon.info glorytun[5679]: STOPPED tun0
Sun Jan 20 12:32:27 2019 daemon.err glorytun[5679]: 51.15.xxxxx.65001: connected
Sun Jan 20 12:32:27 2019 daemon.info glorytun[5679]: STARTED tun0
Sun Jan 20 12:34:24 2019 daemon.err /usr/bin/ss-redir[6803]: send: Broken pipe

Do you have an idea ?

## Specifications

  - OpenMPTCProuter version:  v0.45.15 r0+9007-529c95cc15
  - OpenMPTCProuter VPS version: 0.89
  - OpenMPTCProuter platform: (x86_64) on NUC intel

yoyosm2000, 2019-01-20:
The same here with:
	- openmptcprouter v0.45.15
        - VPS FirstHeberg v0.89

Connection is not stable with a lot of "remote recv: Connection reset by peer" in VPS logs

Ysurac, 2019-01-20:
The "remote recv: Connection reset by peer" means the remote server close the TCP connection with a RESET, which can be caused by timeout or network issue.
For the glorytun problem you should look in VPS logs.

Vinc32du44, 2019-01-20:
How can i look in VPS log  please ?

Ysurac, 2019-01-20:
dmesg and in /var/log

Vinc32du44, 2019-01-20:
Ok thanks Ycarus ;)

Vinc32du44, 2019-02-02:
Again today it sart again ::

`Sat Feb  2 09:35:58 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:35:58 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:35:59 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:35:59 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:04 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:05 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:05 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:06 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:07 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:07 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:08 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:36:08 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:09 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:09 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:16 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:17 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:17 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:18 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:36:18 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:19 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:19 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:20 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:21 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:21 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:23 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:36:23 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:24 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:24 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:25 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:36:25 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:26 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:26 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:27 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:36:28 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:36:28 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:36:44 2019 daemon.err /usr/bin/ss-redir[5729]: remote recv: Connection reset by peer
Sat Feb  2 09:37:35 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:37:35 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:37:36 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:37:36 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:38:07 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:38:07 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:38:08 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:38:08 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:38:09 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:38:10 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:38:10 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:38:12 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:38:13 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:38:13 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:38:14 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:38:14 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:38:15 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:38:15 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:38:58 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:38:59 2019 daemon.err /usr/bin/ss-redir[5729]: remote recv: Connection reset by peer
Sat Feb  2 09:38:59 2019 daemon.err /usr/bin/ss-redir[5729]: remote recv: Connection reset by peer
Sat Feb  2 09:38:59 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:38:59 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:01 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:39:01 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:39:02 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:39:02 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:03 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:39:05 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:39:05 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:06 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:39:07 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:39:07 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:09 2019 daemon.err /usr/bin/ss-redir[5729]: remote recv: Connection reset by peer
Sat Feb  2 09:39:25 2019 daemon.err /usr/bin/ss-redir[5729]: remote recv: Connection reset by peer
Sat Feb  2 09:41:14 2019 daemon.err uhttpd[3794]: luci: accepted login on / for root from 192.168.100.119
[... cut ...]
Sat Feb  2 09:41:25 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:41:26 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:41:26 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:41:27 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:41:27 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:41:28 2019 daemon.err glorytun[4655]: 51.15.235.30.65001: connected
Sat Feb  2 09:41:28 2019 daemon.info glorytun[4655]: STARTED tun0
`
fresh re-install 6 days ago ...

Vinc32du44, 2019-02-02:
It's start again today ::

Sat Feb  2 09:38:59 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:01 2019 daemon.err glorytun[4655]: read: Connection reset by peer
Sat Feb  2 09:39:01 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:39:02 2019 daemon.err glorytun[4655]: 51.15.xxxxx.65001: connected
Sat Feb  2 09:39:02 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:03 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:39:05 2019 daemon.err glorytun[4655]: 51.15.xxxxx.65001: connected
Sat Feb  2 09:39:05 2019 daemon.info glorytun[4655]: STARTED tun0
Sat Feb  2 09:39:06 2019 daemon.info glorytun[4655]: STOPPED tun0
Sat Feb  2 09:39:07 2019 daemon.err glorytun[4655]: 51.15.xxxxx.65001: connected
Sat Feb  2 09:39:07 2019 daemon.info glorytun[4655]: STARTED tun0

Do you have an idea ?

closed 2019-02-02 by Vinc32du44