Ysurac/openmptcprouter #330: OMR-Bypass wan specification

yoyosm2000, 2019-01-20:
Hello, 

I have a lot of connection reset so I want to use OMR-Bypass for one computer on my network, but it  not seems to work properly if I specify wan1 or wan2, it continues to use VPN.

[image]

[image]

[image]

If I choose default, it seems to work...

## Steps to Reproduce the Problem

  1. Specify wan1 or wan2 for an LAN IP in OMR-Bypass: the public IP is the IP of the VPN, it doesn't work
  2. Specify default for the same LAN IP in OMR-Bypass: the public IP is either wan1 or wan2, it works.
  3.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.45.15
  - OpenMPTCProuter VPS version: 0.89
  - OpenMPTCProuter platform: x86_64

Thanks for your help !

Ysurac, 2019-01-20:
This will be fixed in 0.45.16 :)

yoyosm2000, 2019-01-20:
Thanks for your answer and your work !

Ysurac, 2019-01-21:
Should be working in latest release.

closed 2019-01-21 by Ysurac