Ysurac/openmptcprouter #332: Performance graph is chopped

QuaigonSid, 2019-01-23:
My router is ok, all is green and that is a good work! 

However, when i look the performance graph is appeats chopped... It's like russian mountains! 
On my 4g connexion is seem habitual, but my adsl link is anormal. It' seem to follow the graph of the 4g

Is it a performance limitation? Who can i enhance the traffic? 

Thx! 

In french :

Tout fonctionne mais le graph de bande passante fait les montagnes russes sur les deux connections... 
Est ce normal ou peut o  ameliorer les paramètres du routeur pour ameliorer la bande passante ? 

Merci d'avance ! 
## Specifications

  - OpenMPTCProuter version: 45.16
  - OpenMPTCProuter VPS version: Last

  - OpenMPTCProuter platform: rpi 3b+

Ysurac, 2019-01-24:
Don't look at graph :)
How do you test speed for that ?

QuaigonSid, 2019-01-25:
This graph :    
[image]. 

So the mptcp is operational and i just want to know if this graph is "normal". 
I have 2 connections adsl and 4g and the adsl is the master. 

Thanks