Ysurac/openmptcprouter #333: power outage/ readonly

tech1Data, 2019-01-23:
When i'm truning off the raspberry during the reboot file system is in read-only that's why i have find a solution with e2fck but when i'm launching a script in startup he is not working.  that's the script:
```bash
#! /bin/bash
touch /root/canWrite
if [ ! -f "/root/canWrite" ]; then
e2fsck -y /dev/mmcblk0p2
reboot
else
rm /root/canWrite
fi
```
the name of the script is /root/script.sh

in the rc.local i put the following line : 
```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
#!/bin/sh -e
sh /root/script.sh
exit 0
```
but not working
## Steps to Reproduce the Problem

  1. Write script
  2.Put start script in rc.local
  3.simulate power outage until the read-only filesystem

## Specifications
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

lars18th, 2019-02-13:
Hi,

For all devices using one SD card for the filesystem, I recommend to not put the entire filesystem inside the same partition. All current SD cards have a very low write lifespan.

So. I suggest to support a more "complex" but more robust filesystem, like:
- One RO squashfs for the filesystem.
- One RW overlay over the SD card (or any other partition).
- One RW temp filesystem in RAM.
- Plus, auto-fsck of the RW overlay at _each_ boot.
- Plus, a specific filesystem format for the SD card (f2fs, or ext4 without journal).

This is the common solution with industrial devices.

closed 2019-07-04 by Ysurac

lars18th, 2019-07-05:
Hi,

I have killed dozens of SD cards for using it as root filesystems for several devices. So I strongly recommend using a root filesystem over RAM and use overlays. That's the common behaviour of OpenWrt, but not the common configuration for the Pi's.

Regards.

Actpohomoc, 2021-04-28:
> Hi,
> 
> I have killed dozens of SD cards for using it as root filesystems for several devices. So I strongly recommend using a root filesystem over RAM and use overlays. That's the common behaviour of OpenWrt, but not the common configuration for the Pi's.
> 
> Regards.

How to do it for Pi?

Actpohomoc, 2021-04-28:
There is no solution for the topic?
* Using other device not PI?
* Using power backup with graceful shutdown?