Ysurac/openmptcprouter #335: mptcp_detect_mapping Mappings do not match!

kapook007, 2019-01-24:
[118462.591379] mptcp_detect_mapping Mappings do not match!
[118462.596720] mptcp_detect_mapping dseq 2579996328 mdseq 2579996328, sseq 1741402433 msseq 1741399677 dlen 10969 mdlen 10968 dfin 1 mdfin 1
[118463.109901] mptcp_detect_mapping Mappings do not match!
[118463.115242] mptcp_detect_mapping dseq 2579996328 mdseq 2579996328, sseq 1743303461 msseq 1743300705 dlen 10969 mdlen 10968 dfin 1 mdfin 1

mptcprouter0.45.12
vps0.89

Ysurac, 2019-01-24:
I would need more info. Connections used ? (4g, adsl, ffth,...) Congestion control used ? how often you have the error ?...

kapook007, 2019-01-25:
internet connections is 2x ftth .
Congestion control is bbr.
error when download tcp over 1hr.

Ysurac, 2019-02-05:
You should try latest release.
If no change check for VPS and router resources usages.

closed 2019-03-13 by Ysurac

xabix, 2019-05-21:
Hello,

I have the same issue with the latest release: openmptcprouter v0.49.5 r0+9945-bc85640cdc

What can I info should I be looking to better troubleshoot the issue?
[image]

Merci
XabiX