Ysurac/openmptcprouter #337: OMR-Bypass - Failed to execute cbi dispatcher target for entry '/admin/services/omr-bypass/index'

G000R000, 2019-01-25:
## Expected Behavior

Using the OMR-Bypass service (menu Services  > OMR-Bypass)

## Actual Behavior

Having the following error :

Failed to execute cbi dispatcher target for entry '/admin/services/omr-bypass/index'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: bad argument #1 to 'lines' (/proc/net/xt_ndpi/proto: No such file or directory)
stack traceback:
	[C]: in function 'lines'
	/usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:838: in function </usr/lib/lua/luci/dispatcher.lua:832>

## Steps to Reproduce the Problem

  1. Go the menu Services
  2.Click OMR-Bypass
  3. error appears

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.45.16 r0+9007-529c95cc15 / LuCI Master (git-19.020.70956-5b0097b) > i fresh installed (maybe 0.45 ?) then upgrade 2 times with 0.45.15 and 0.45.16
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.89
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2

Ysurac, 2019-01-25:
The problem is the read only filesystem. You should write a new image.

Ysurac, 2019-01-25:
Wipe everything with a new image. Upgrade will not work well.

G000R000, 2019-01-25:
Did a full reinstall on v0.45.16 than restaured my backup. It worked, thanks Yannick

closed 2019-01-25 by G000R000