Ysurac/openmptcprouter #338: OMR By-pass error

dnwk, 2019-01-26:
Failed to execute cbi dispatcher target for entry '/admin/services/omr-bypass/index'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: bad argument #1 to 'lines' (/proc/net/xt_ndpi/proto: No such file or directory)
stack traceback:
	[C]: in function 'lines'
	/usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:838: in function </usr/lib/lua/luci/dispatcher.lua:832>

## Specifications

  - OpenMPTCProuter version: 0.45.16 r0+9007-529c95cc15
  - OpenMPTCProuter VPS version: 0.88
  - OpenMPTCProuter platform: Linksys WRT32X

Ysurac, 2019-01-26:
Check openmptcprouter logs for ndpi.

tukutt, 2019-01-27:
Same things here with RPI 2B.

Sat Jan 26 15:06:19 2019 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
Sat Jan 26 15:06:23 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sat Jan 26 15:06:24 2019 daemon.err omr-6in4[4060]: Command failed: Not found
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: Failed to execute cbi dispatcher target for entry '/admin/services/omr-bypass/index'.
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: The called action terminated with an exception:
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: /usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: bad argument #1 to 'lines' (/proc/net/xt_ndpi/proto: No such file or directory)
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: stack traceback:
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: 	[C]: in function 'lines'
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: 	/usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: in function 'func'
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: 	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
Sat Jan 26 15:23:20 2019 daemon.err uhttpd[2003]: 	/usr/lib/lua/luci/dispatcher.lua:838: in function </usr/lib/lua/luci/dispatcher.lua:832`

dnwk, 2019-01-27:
@tukutt  could you check the logs as @Ysurac  mentioned? I already revert back to old version.

tukutt, 2019-01-27:
@dnwk Where I can found this information ? Thx

Ysurac, 2019-01-27:
In Status->Kernel log and System log

tukutt, 2019-01-27:
Kernel log :
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.14.91 (circleci@default-aaced3fb-866a-412f-91b6-70da4cff4abb) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r0+9007-529c95cc15)) #0 SMP Mon Jan 7 16:13:59 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 16 MiB at 0x3a400000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80927ac0, node_mem_map b9c88000
[    0.000000]   Normal zone: 1896 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x84/0x360 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @b9c36000 s33676 r8192 d23668 u65536
[    0.000000] pcpu-alloc: s33676 r8192 d23668 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240792
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 936808K/970752K available (5120K kernel code, 160K rwdata, 1284K rodata, 1024K init, 365K bss, 17560K reserved, 16384K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80600000   (6112 kB)
[    0.000000]       .init : 0x80800000 - 0x80900000   (1024 kB)
[    0.000000]       .data : 0x80900000 - 0x809282c0   ( 161 kB)
[    0.000000]        .bss : 0x8092d30c - 0x80988a68   ( 366 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000259] Console: colour dummy device 80x30
[    0.000787] console [tty1] enabled
[    0.000837] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000881] pid_max: default: 32768 minimum: 301
[    0.001055] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001091] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001790] CPU: Testing write buffer coherency: ok
[    0.002288] CPU0: update cpu_capacity 1024
[    0.002320] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.002798] Setting up static identity map for 0x100000 - 0x100060
[    0.002970] Hierarchical SRCU implementation.
[    0.003752] smp: Bringing up secondary CPUs ...
[    0.004639] CPU1: update cpu_capacity 1024
[    0.004648] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.005588] CPU2: update cpu_capacity 1024
[    0.005597] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.006469] CPU3: update cpu_capacity 1024
[    0.006478] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.006589] smp: Brought up 1 node, 4 CPUs
[    0.006710] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006731] CPU: All CPU(s) started in HYP mode.
[    0.006749] CPU: Virtualization extensions available.
[    0.007970] devtmpfs: initialized
[    0.018635] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.018967] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.019016] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.020025] pinctrl core: initialized pinctrl subsystem
[    0.020455] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.021213] NET: Registered protocol family 16
[... 521 lines cut ...]
[   11.262641] usbcore: registered new interface driver rt2800usb
[   11.282179] usbcore: registered new interface driver rtl8192cu
[   11.289680] kmodloader: 1 module could not be probed
[   11.296398] kmodloader: - xt_ndpi - 0
[   15.581827] random: crng init done
[   15.586873] random: 6 urandom warning(s) missed due to ratelimiting
[   17.033219] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   18.620758] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[... cut ...]
Sun Jan 27 21:45:40 2019 daemon.err uhttpd[2003]: stack traceback:
Sun Jan 27 21:45:40 2019 daemon.err uhttpd[2003]: 	[C]: in function 'lines'
Sun Jan 27 21:45:40 2019 daemon.err uhttpd[2003]: 	/usr/lib/lua/luci/model/cbi/omr-bypass.lua:90: in function 'func'
Sun Jan 27 21:45:40 2019 daemon.err uhttpd[2003]: 	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
Sun Jan 27 21:45:40 2019 daemon.err uhttpd[2003]: 	/usr/lib/lua/luci/dispatcher.lua:838: in function </usr/lib/lua/luci/dispatcher.lua:832>

Ysurac, 2019-01-27:
It's à bug for 32bits arm in ndpi-netfilter. This Will be fixed with next release.

Ysurac, 2019-02-05:
Fixed in latest release.

closed 2019-02-05 by Ysurac