Ysurac/openmptcprouter #339: Can't Get Public IP address from shadowSocks

oOSphaXOo, 2019-01-26:
## Expected Behavior

 Can't Get Public IP address from shadowSocks. I have try to reinstall and manually reconfigure (no importation file ) but same bug ...

[image]

> Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624411] pinctrl-bcm2835 3f200000.gpio: read 00000000 (32 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624416] pinctrl-bcm2835 3f200000.gpio: write 000001c0 (32 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624424] pinctrl-bcm2835 3f200000.gpio: request pin 33 (gpio33) for 3f201000.serial
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624429] pinctrl-bcm2835 3f200000.gpio: read 000001c0 (33 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624433] pinctrl-bcm2835 3f200000.gpio: write 00000fc0 (33 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624440] pinctrl-bcm2835 3f200000.gpio: request pin 43 (gpio43) for 3f201000.serial
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.624444] pinctrl-bcm2835 3f200000.gpio: read 00200924 (43 => alt0)
Sat Jan 26 20:46:08 2019 kern.info kernel: [    0.624505] uart-pl011 3f201000.serial: cts_event_workaround enabled
Sat Jan 26 20:46:08 2019 kern.info kernel: [    0.625449] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 72, base_baud = 0) is a PL011 rev2
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.627793] pinctrl core: add 0 pinctrl maps
Sat Jan 26 20:46:08 2019 kern.info kernel: [    0.628008] console [ttyS0] disabled
Sat Jan 26 20:46:08 2019 kern.info kernel: [    0.628888] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 151, base_baud = 31250000) is a 16550
Sat Jan 26 20:46:08 2019 kern.warn kernel: [    0.801105] Indeed it is in host mode hprt0 = 00021501
Sat Jan 26 20:46:08 2019 kern.info kernel: [    0.803458] console [ttyS0] enabled
Sat Jan 26 20:46:08 2019 kern.notice kernel: [    0.916379] random: fast init done
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    0.920250] pinctrl core: add 12 pinctrl maps
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.011061] usb 1-1: new high-speed USB device number 2 using dwc_otg
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.013415] pinctrl-bcm2835 3f200000.gpio: found group selector 34 for gpio34
Sat Jan 26 20:46:08 2019 kern.warn kernel: [    1.019831] Indeed it is in host mode hprt0 = 00001101
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.025064] pinctrl-bcm2835 3f200000.gpio: found group selector 35 for gpio35
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.271867] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.276689] pinctrl-bcm2835 3f200000.gpio: found group selector 36 for gpio36
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.282327] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.282621] hub 1-1:1.0: no of_node; not parsing pinctrl DT
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.285698] pinctrl-bcm2835 3f200000.gpio: found group selector 37 for gpio37
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.592496] pinctrl-bcm2835 3f200000.gpio: found group selector 38 for gpio38
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.592507] pinctrl-bcm2835 3f200000.gpio: found group selector 39 for gpio39
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.592520] pinctrl-bcm2835 3f200000.gpio: request pin 34 (gpio34) for 3f300000.mmc
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.592526] pinctrl-bcm2835 3f200000.gpio: read 00000fc0 (34 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.592531] pinctrl-bcm2835 3f200000.gpio: write 00007fc0 (34 <= alt3)
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.592535] hub 1-1:1.0: USB hub found
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.592540] pinctrl-bcm2835 3f200000.gpio: request pin 35 (gpio35) for 3f300000.mmc
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.597191] pinctrl-bcm2835 3f200000.gpio: read 00007fc0 (35 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.597195] pinctrl-bcm2835 3f200000.gpio: write 0003ffc0 (35 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.597202] pinctrl-bcm2835 3f200000.gpio: request pin 36 (gpio36) for 3f300000.mmc
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.597207] pinctrl-bcm2835 3f200000.gpio: read 0003ffc0 (36 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.597211] pinctrl-bcm2835 3f200000.gpio: write 001fffc0 (36 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.597262] pinctrl-bcm2835 3f200000.gpio: request pin 37 (gpio37) for 3f300000.mmc
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.597264] hub 1-1:1.0: 5 ports detected
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602161] pinctrl-bcm2835 3f200000.gpio: read 001fffc0 (37 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602165] pinctrl-bcm2835 3f200000.gpio: write 00ffffc0 (37 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602172] pinctrl-bcm2835 3f200000.gpio: request pin 38 (gpio38) for 3f300000.mmc
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602177] pinctrl-bcm2835 3f200000.gpio: read 00ffffc0 (38 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602181] pinctrl-bcm2835 3f200000.gpio: write 07ffffc0 (38 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602187] pinctrl-bcm2835 3f200000.gpio: request pin 39 (gpio39) for 3f300000.mmc
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602192] pinctrl-bcm2835 3f200000.gpio: read 07ffffc0 (39 => gpio_in)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602196] pinctrl-bcm2835 3f200000.gpio: write 3fffffc0 (39 <= alt3)
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602363] mmc-bcm2835 3f300000.mmc: GPIO lookup for consumer wp
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602367] mmc-bcm2835 3f300000.mmc: using device tree for GPIO lookup
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602374] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node '/soc/mmc@7e300000[0]'
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602379] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node '/soc/mmc@7e300000[0]'
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602383] mmc-bcm2835 3f300000.mmc: using lookup tables for GPIO lookup
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.602387] mmc-bcm2835 3f300000.mmc: lookup for GPIO wp failed
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.602416] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
Sat Jan 26 20:46:08 2019 kern.info kernel: [    1.609163] mmc-bcm2835 3f300000.mmc: DMA channel allocated
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671154] pinctrl core: add 6 pinctrl maps
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671168] pinctrl-bcm2835 3f200000.gpio: found group selector 48 for gpio48
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671176] pinctrl-bcm2835 3f200000.gpio: found group selector 49 for gpio49
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671184] pinctrl-bcm2835 3f200000.gpio: found group selector 50 for gpio50
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671191] pinctrl-bcm2835 3f200000.gpio: found group selector 51 for gpio51
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671199] pinctrl-bcm2835 3f200000.gpio: found group selector 52 for gpio52
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671206] pinctrl-bcm2835 3f200000.gpio: found group selector 53 for gpio53
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671213] pinctrl-bcm2835 3f200000.gpio: request pin 48 (gpio48) for 3f202000.mmc
Sat Jan 26 20:46:08 2019 kern.debug kernel: [    1.671218] pinctrl-bcm2835 3f200000.gpio: read 00200924 (48 => gpio_in)
[... 1166 lines cut ...]
Sat Jan 26 21:03:09 2019 daemon.err glorytun[22600]: 51.77.XXX.XXX.65001: key exchange failed
Sat Jan 26 21:03:09 2019 daemon.info glorytun[22600]: STOPPED tun0
Sat Jan 26 21:03:11 2019 daemon.err glorytun[22600]: 51.77.XXX.XXX.65001: connected
Sat Jan 26 21:03:21 2019 daemon.err glorytun[22600]: 51.77.XXX.XXX.65001: key exchange failed
Sat Jan 26 21:03:21 2019 daemon.info glorytun[22600]: STOPPED tun0
Sat Jan 26 21:03:22 2019 daemon.err glorytun[22600]: 51.77.XXX.XXX.65001: connected
Sat Jan 26 21:03:32 2019 daemon.err glorytun[22600]: 51.77.XXX.XXX.65001: key exchange failed
Sat Jan 26 21:03:32 2019 daemon.info glorytun[22600]: STOPPED tun0
Sat Jan 26 21:03:33 2019 daemon.err glorytun[22600]: 51.77.XXX.XXX.65001: connected

## Actual Behavior

## Steps to Reproduce the Problem

  1. VPS OVH
  2. Install from Scratch on PI
  3. Configure
  4. Works a couple of hours
  5. Wait ...  and Bug 

## Specifications

  - OpenMPTCProuter version: 0.45.16
  - OpenMPTCProuter VPS version: 0.89
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-01-27:
You should disable UPnP (because this should always be disabled ;) ).
You can also disable IPv6 if enabled.

oOSphaXOo, 2019-01-27:
Thanks I try. So 2 another question:
- Why UPnP function is implemented if it must be always disable ^_^ ?
- Periphericals with UPnP (like PS4 / One) must be configurated manualy (without UPnp) ?

Ysurac, 2019-01-27:
* Because some users really want this. Even if it's a really bad idea and a big security risk when enabled.
* I think so, but no idea. I don't have anything that can use UPnP ;)

oOSphaXOo, 2019-01-27:
[uci.show.txt](https://github.com/Ysurac/openmptcprouter/files/2799885/uci.show.txt)

oOSphaXOo, 2019-01-27:
BBR activated (when I go in MPTCP Interface "Congestion Control" was on "olia" if this can help you to debug).
When I re-select "bbr" i have "network.globals.congestion='bbr'" (with ssh and "uci show | grep congestion" command)
OpenMPTCPRouter look fine since 1 hour. I hope that is the solution ^^

oOSphaXOo, 2019-01-27:
No Issue since 9 hours. I think that was the solution 👍 
Thanks for your work and your help. You and OpenMPTCPRouter are top !!!
Un voisAIN ^^