Ysurac/openmptcprouter #342: Ftp disconnect

QuaigonSid, 2019-02-02:
## Expected Behavior

Ftp download a entire file with no disconnect

## Actual Behavior

Ftp disconnect after some time of downloading 
(file bigher than 1go)

Error log : mptcp_detect_mapping Mappings do not match!

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: Last
  - OpenMPTCProuter VPS version: Last
  - OpenMPTCProuter platform: rpi 3 b+

    Wan1 : free adsl
     Wan2 : bouygues 4g

  Is the random latency of the 4g the base of the problem?

Ysurac, 2019-02-05:
You should latest release, v0.46.1

QuaigonSid, 2019-02-06:
Okay! I will...
It's version 0.46.2 and 0.90 for the vps. 

I will try ftp connection!

++

QuaigonSid, 2019-02-24:
Nope... Dosen't work anymore... Always the same error : mapping doesn't match...

Ysurac, 2019-02-24:
I'm not able to reproduce the error.
Can you give me the result of ```uci show network``` via SSH on OpenMPTCProuter ?

closed 2019-03-13 by Ysurac