Ysurac/openmptcprouter #343: Baux dhcp statique

Dragonox77, 2019-02-03:
## Expected Behavior
Bonjour, merci pour le travail accompli et désolé si mon issue n'est pas rédigée en anglais.
 Les baux DHCP statiques devraient fonctionner.

## Actual Behavior

Quand je configure un bail dhcp static, il ne fonctionne tout simplement pas.
## Steps to Reproduce the Problem

  1. Creer un bail dhcp static
  2. Se déconnecter et reconnecter le périphérique que vous avez configurer
  3. Voir que le bail dhcp n'est pas actif et ne fonctionne pas.

## Specifications

  - OpenMPTCProuter version: 0.45.16
  - OpenMPTCProuter VPS version: 0.89
  - OpenMPTCProuter platform: RPI3 x64

Ysurac, 2019-02-03:
On peut déjà changer la passerelle par défaut et sinon le bail DHCP static fonctionne correctement.

Dragonox77, 2019-02-03:
Avez vous testé cette fonctionnalité ? Parce-que pour ma part, il n'y a aucun moyen de faire fonctionner ça chez moi.
Nous sommes bien d'accord qu'il peut y avoir des baux dhcp statique quand le serveur dhcp basique est activé ?

Peut être que je n'ai absolument pas compris comment ça fonctionne malgré toutes mes recherches sur le net. Je veux juste qu'un périphérique puisse bénéficier d'une ip local statique avec une passerelle autre que celle d'openmptcprouter.

[image]

L'adresse ip configurée est 192.168.1.17, aucun périphérique ne détient cette ip.
Ai-je bien rempli tous les champs nécessaire ?

J'espère vraiment pas vous embêter avec mon problème même si il est très important pour moi de le régler.

Je vous remercie pour le temps que vous voudriez bien apporter à votre réponse.

PxxBxx, 2019-02-04:
Bonjour @Dragonox77 peut être est-ce parce que votre OMR est configuré (par défaut) sur le subnet 192.168.100.0/24 ?
Si c'est le cas, c'est normal que le bail en 192.168.1.0/24 ne puisse être propagé (on ne peut pas "dhcp leaser" une IP en dehors du subnet du serveur DHCP).

(j'ai essayé à l'époque, mais je n'ai pas cherché plus que ça, il faudrait ajouter manuellement des subnets au dhcp rien que pour ça)

Dragonox77, 2019-02-04:
Bonjour,
Merci pour vos réponses.

Après avoir regarder des forums de discutions, j'ai fini par ajouter ces quelques lignes au fichier de configuration de dnsmasq:
dhcp-range=net:2,192.168.1.0,255.255.255.0,static
dhcp-option=net:2,3,192.168.1.1

dhcp-host=2C:56:XX:XX:XX:XX,net:2,192.168.1.73

Sauf que ....... ca marche toujours pas... pourtant tout me parait bon, je comprend pas.

ThibautBascoulergue, 2019-02-04:
Comme le dis PxxBxx , si tu n'es pas dans le même subnet, ton dhcp ne pourra donner une ip qui n'appartient pas à ce subnet :/
En gros si ton mptcp est en 192.168.100.1 , tu pourra donner des ip dans le range 192.168.100.100-253.

Dragonox77, 2019-02-04:
Pourriez vous me dire la marche a suivre pour que dans mon cas cela fonctionne ?

Ysurac, 2019-02-04:
@Dragonox77 si OpenMPTCProuter est en 192.168.100.1, le serveur DHCP ne peut pas donner une IP en 192.168.1.x. Mais rien empêche d'ajouter une interface LAN en 192.168.1.x.

Dragonox77, 2019-02-04:
Merci mille fois, tout fonctionne parfaitement !

Merci merci.

closed 2019-02-04 by Dragonox77