Ysurac/openmptcprouter #344: cannot update packages on openmptcprouter v0.45.16

cth0, 2019-02-03:
## Expected Behavior
Hello,

I upgraded OpenMPTCPRouter to v0.45.16 tonight.
Everything ran fine, but after upgrade,  I wanted to reinstall some packages for monitoring (softflowd).

Running the "opkg update" command fails with thes errrors : 

 * opkg_download: Failed to download http://downloads.openwrt.org/releases/18.06.0/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/releases/18.06.0/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/releases/18.06.0/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/releases/18.06.0/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/releases/18.06.0/packages/aarch64_cortex-a53/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

Other packages are correctly dowloaded.

I'm able to download packages directly from my laptop, but not from the router.
Trying to directly download results in : 

Downloading 'http://downloads.openwrt.org/releases/18.06.0/packages/aarch64_cortex-a53/luci/Packages.gz'
Connecting to 2a01:4f8:202:43ea::2:80
Connection error: Connection failed

Any clue on this?

Config details: 
- OpenMPTCPRouter v0.45.16
- MPS server v.82

Ysurac, 2019-02-03:
IPv6 is enabled ? Seems to try to connect using IPv6 address.

cth0, 2019-02-03:
yes that was my feeling.

I disabled IPv6 in /etc/sysctl.d/zzz_openmptcprouter.conf : 

```
net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6 = 1

```

and now the update went fine!

closed 2019-02-05 by Ysurac