Ysurac/openmptcprouter #345: Upload speed

zicochaos, 2019-02-05:
## Expected Behavior

Upload speed via both interfaces 

## Actual Behavior

Upload is going via one interface 

## Steps to Reproduce the Problem

  1. Any speed test 
  
 

## Specifications

  - OpenMPTCProuter version:  v0.45.16 r0+9007-529c95cc15
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: x86_64)

[image]

Ysurac, 2019-02-05:
Speed tests are not really good to check upload speed with MPTCP. You should try by uploading some real data.

zicochaos, 2019-02-06:
Same result. Is any way to choose interface for upload?

closed 2019-02-12 by zicochaos