Ysurac/openmptcprouter #349: slow speed, then VPN down

olaulau, 2019-02-08:
## Expected Behavior
acceptable speed, stable connection

## Actual Behavior
download speed is very slow
after some time (about 10 min), VPN disconnects

## Steps to Reproduce the Problem
  1.install OMR vps and router
  2.use wizard to configure stuff

## Specifications
  - OpenMPTCProuter version: 0.46.2
  - OpenMPTCProuter VPS version: 0.92
  - OpenMPTCProuter platform: RPI3
fresh install from this morning (both VPS and RPI).
VPS is on a VM on a OVH dedicated host with an failover IP
RPI model is 3B+, no power problem, I have a screen attached to watch.
standart install with wizard, only one Free ADSL connection for the moment (about 19Mbps), connected via rpi ethernet port, and a switch.

extract from syslog :
`Fri Feb  8 10:23:50 2019 daemon.info dnsmasq[1274]: read /etc/hosts - 4 addresses
Fri Feb  8 10:23:50 2019 daemon.info dnsmasq[1274]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Feb  8 10:23:50 2019 user.notice omr-bypass: OMR-ByPass is running
Fri Feb  8 10:23:51 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Fri Feb  8 10:23:52 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Fri Feb  8 10:23:54 2019 user.notice post-tracking-post-tracking: omrvpn down
Fri Feb  8 10:30:05 2019 authpriv.info dropbear[12972]: Child connection from 192.168.100.2:51290
Fri Feb  8 10:30:07 2019 authpriv.notice dropbear[12972]: Password auth succeeded for 'root' from 192.168.100.2:51290
Fri Feb  8 10:30:43 2019 daemon.info glorytun[9036]: STOPPED tun0
Fri Feb  8 10:31:03 2019 daemon.err /usr/bin/ss-local[9837]: getpeername: Socket not connected
Fri Feb  8 10:31:24 2019 daemon.err /usr/bin/ss-local[9837]: getpeername: Socket not connected
Fri Feb  8 10:31:44 2019 daemon.err /usr/bin/ss-local[9837]: getpeername: Socket not connected
Fri Feb  8 10:31:58 2019 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 103.224.182.242, 198.27.92.1, 88.191.250.176, 151.101.129.164)
Fri Feb  8 10:31:58 2019 user.notice Shadowsocks: Rules DOWN
Fri Feb  8 10:32:02 2019 daemon.err omr-tracker-ss[11220]: Failed to establish connection
Fri Feb  8 10:32:04 2019 daemon.err /usr/bin/ss-local[9837]: getpeername: Socket not connected
Fri Feb  8 10:32:17 2019 daemon.err /usr/bin/ss-local[9837]: getpeername: Socket not connected
Fri Feb  8 10:32:26 2019 daemon.err /usr/bin/ss-redir[9841]: remote recv: Operation timed out
Fri Feb  8 10:32:27 2019 daemon.err /usr/bin/ss-redir[9840]: remote recv: Operation timed out`

nothing particular in kernel log :
`[   13.310376] random: crng init done
[   13.317054] random: 6 urandom warning(s) missed due to ratelimiting
[   14.014167] 8021q: adding VLAN 0 to HW filter on device eth0
[   14.015148] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   14.035162] IPv6: ADDRCONF(NETDEV_UP): wan1: link is not ready
[   14.044233] 8021q: adding VLAN 0 to HW filter on device wan1
[   15.111201] IPv6: ADDRCONF(NETDEV_CHANGE): wan1: link becomes ready
`

network overview :
[image]

olaulau, 2019-02-08:
don't loose time with this, I've reset the router config and done the wizard again.
It's working great at this time ! I'll reopen if problem occurs again.

fyi, here is a comparision of speedtest done via speedtest.net in browser, with OMR and without (on my single DSL connection) :

| | latency (ms) | download (mbps) | upload (mbps) |
| --- | --- | --- | --- |
| normal | 9 | 18 | 0.83 |
| OMR | 18 | 16.5 | 0.77 |

very little loss in this case, you've done a great job !
thanks.

olaulau, 2019-02-08:
problems never leave from themselves !
VPN down again :

syslog
```
Fri Feb  8 14:57:41 2019 daemon.info glorytun[23839]: STOPPED tun0
Fri Feb  8 14:57:42 2019 daemon.err glorytun[23839]: 178.33.221.169.65001: connected
Fri Feb  8 14:57:42 2019 daemon.info glorytun[23839]: STARTED tun0
Fri Feb  8 15:38:49 2019 daemon.err /usr/bin/ss-redir[26683]: remote recv: Operation timed out
Fri Feb  8 15:38:49 2019 daemon.err /usr/bin/ss-redir[26682]: remote recv: Operation timed out
Fri Feb  8 15:38:49 2019 daemon.err glorytun[23839]: read: Operation timed out
Fri Feb  8 15:38:49 2019 daemon.info glorytun[23839]: STOPPED tun0
Fri Feb  8 15:39:10 2019 daemon.err /usr/bin/ss-local[26680]: getpeername: Socket not connected
Fri Feb  8 15:39:16 2019 daemon.err glorytun[23839]: 178.33.221.169.65001: connected
Fri Feb  8 15:39:16 2019 daemon.info glorytun[23839]: STARTED tun0
Fri Feb  8 15:39:39 2019 daemon.err glorytun[23839]: read: Operation timed out
Fri Feb  8 15:39:39 2019 daemon.info glorytun[23839]: STOPPED tun0
Fri Feb  8 15:39:40 2019 daemon.err glorytun[23839]: 178.33.221.169.65001: connected
Fri Feb  8 15:39:40 2019 daemon.info glorytun[23839]: STARTED tun0
Fri Feb  8 15:39:40 2019 daemon.err /usr/bin/ss-local[26680]: getpeername: Socket not connected
Fri Feb  8 15:39:55 2019 daemon.err glorytun[23839]: read: Operation timed out
Fri Feb  8 15:39:55 2019 daemon.info glorytun[23839]: STOPPED tun0
Fri Feb  8 15:40:00 2019 daemon.err /usr/bin/ss-local[26680]: getpeername: Socket not connected
Fri Feb  8 15:40:07 2019 daemon.err glorytun[23839]: 178.33.221.169.65001: connected
Fri Feb  8 15:40:07 2019 daemon.info glorytun[23839]: STARTED tun0
Fri Feb  8 15:40:31 2019 daemon.err glorytun[23839]: read: Operation timed out
Fri Feb  8 15:40:31 2019 daemon.info glorytun[23839]: STOPPED tun0
Fri Feb  8 15:40:33 2019 daemon.err /usr/bin/ss-local[26680]: getpeername: Socket not connected
Fri Feb  8 15:40:44 2019 daemon.err glorytun[23839]: 178.33.221.169.65001: connected
Fri Feb  8 15:40:44 2019 daemon.info glorytun[23839]: STARTED tun0
```
nothing in kernel logs.

olaulau, 2019-02-08:
after a router reboot, all is working again :-/

olaulau, 2019-02-08:
VPN down again :
```
Fri Feb  8 15:48:59 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Fri Feb  8 15:49:28 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Fri Feb  8 15:55:10 2019 daemon.err glorytun[3147]: read: Operation timed out
Fri Feb  8 15:55:10 2019 daemon.info glorytun[3147]: STOPPED tun0
Fri Feb  8 15:55:11 2019 daemon.err glorytun[3147]: 178.33.221.169.65001: connected
Fri Feb  8 15:55:11 2019 daemon.info glorytun[3147]: STARTED tun0
Fri Feb  8 15:55:13 2019 daemon.err /usr/bin/ss-redir[4289]: remote recv: Operation timed out

```

Ysurac, 2019-02-08:
There is a timeout on the VPN and a few seconds later with shadowsocks, so this seems to be a problem between master wan and the server. If it's a scaleway VPS they have an issue today.

olaulau, 2019-02-11:
the server hosting OMR VPS part is a dedicated host at SoYouStart (OVH).

Ysurac, 2019-02-11:
Can be a problem with the wan used, or a wrong configuration.

closed 2019-03-13 by Ysurac