Ysurac/openmptcprouter #350: Catastrophic rate since last update

Roken120, 2019-02-09:
## Expected Behavior

I have Catastrophic rate since last update

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 10.0G    3  355M    0     0  10.4M      0  0:16:19  0:00:34  0:15:45 7544k^C
root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 70.9M    0     0  8047k      0  0:21:43  0:00:09  0:21:34 11.0M^C
root@OpenMPTCProuter:~# omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 79.5M    0     0  10.4M      0  0:16:16  0:00:07  0:16:09 11.2M^C
root@OpenMPTCProuter:~# omr-test-speed wan3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 25.4M    0     0  3385k      0  0:51:36  0:00:07  0:51:29 3476k^C

## Actual Behavior

## Steps to Reproduce the Problem

  1.Update to last update
  2.Speed test

## Specifications

  - OpenMPTCProuter version: 0.46.3
  - OpenMPTCProuter VPS version: 0.92
  - OpenMPTCProuter platform: x86_64

Roken120, 2019-02-09:
I have reinstall my system and i have same error

Ysurac, 2019-02-09:
I need the config. ```uci show``` via SSH (remove server ip settings)

Roken120, 2019-02-09:
uci show of router : https://pastebin.com/bUT4y6Jc

Ysurac, 2019-02-09:
Lan, WAN1 and  WAN2 are on the same network. This can't work that way.

Roken120, 2019-02-09:
Its working before why now it works more

Ysurac, 2019-02-09:
No idea how this can work ;)

Roken120, 2019-02-09:
Ok maybe what is the bug ? because my rate speed is max 70mbps...

Ysurac, 2019-02-09:
There is no bug here.
Lan, and WANs MUST be on separated networks.
Else MPTCP have no idea what the route it can use for each packets.

Roken120, 2019-02-09:
Ok i change

Envoyé le :samedi 9 février 2019 16:49
À : Ysurac/openmptcprouter<mailto:openmptcprouter@noreply.github.com>
Cc : Roken120<mailto:pumadu@hotmail.fr>; Author<mailto:author@noreply.github.com>

Roken120, 2019-02-12:
I have change and i have same result

Ysurac, 2019-02-12:
Can I have the result of ```uci show``` now that changes are made ?

closed 2019-03-13 by Ysurac