Ysurac/openmptcprouter #351: MPTCProuter 0.46.3  Glorytun tcp disconnet every 2 hr.

kapook007, 2019-02-09:
MPTCProuter with Glorytun tcp .
forward port disconnet every 2 hr.
OMRVPN interface  dhcp Lease Time is 2Hr.
can not change OMRVPN  interface to static ip.

MPTCProuter 0.46.3

Ysurac, 2019-02-09:
The name is OpenMPTCProuter.
What is the VPS version ?
DHCP is given for 2 hours but should not disconnect for that...
Not enough log to check where is the problem.

kapook007, 2019-02-09:
Linksys WRT3200ACM  with MPTCProuter 0.46.3
VPS is 0.92

i down again  and save log

kapook007, 2019-02-09:
Sat Feb  9 22:00:00 2019 cron.info crond[2841]: USER root pid 4383 cmd /etc/init.d/vnstat_backup backup
Sat Feb  9 22:00:00 2019 user.info vnstat_backup: backing up database
Sat Feb  9 22:14:57 2019 daemon.err /usr/bin/ss-local[8348]: getpeername: Socket not connected
Sat Feb  9 22:29:44 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Sat Feb  9 22:30:07 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sat Feb  9 22:31:09 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Sat Feb  9 22:31:25 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sat Feb  9 22:40:40 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 10.255.255.1
Sat Feb  9 22:45:57 2019 daemon.err /usr/bin/ss-redir[8352]: remote recv: Host is unreachable
Sat Feb  9 23:08:24 2019 daemon.err /usr/bin/ss-redir[8351]: remote recv: Host is unreachable
Sat Feb  9 23:10:40 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 10.255.255.1
Sat Feb  9 23:25:40 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 10.255.255.1
Sat Feb  9 23:30:08 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Sat Feb  9 23:30:25 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sat Feb  9 23:31:32 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Sat Feb  9 23:31:48 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sat Feb  9 23:33:10 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 10.255.255.1
Sat Feb  9 23:36:55 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 10.255.255.1
Sat Feb  9 23:38:47 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 10.255.255.1
Sat Feb  9 23:39:43 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 0.0.0.0
Sat Feb  9 23:40:12 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 0.0.0.0
Sat Feb  9 23:40:26 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 0.0.0.0
Sat Feb  9 23:40:34 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 0.0.0.0
Sat Feb  9 23:40:37 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 0.0.0.0
Sat Feb  9 23:40:38 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending renew to 0.0.0.0
Sat Feb  9 23:40:38 2019 daemon.notice netifd: omrvpn (6546): udhcpc: lease lost, entering init state
Sat Feb  9 23:40:38 2019 daemon.notice netifd: Interface 'omrvpn' has lost the connection
Sat Feb  9 23:40:39 2019 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.18.51.1 dev eth2.101
Sat Feb  9 23:40:40 2019 user.notice omr-bypass: Starting OMR-ByPass...
Sat Feb  9 23:40:40 2019 daemon.info dnsmasq[10813]: read /etc/hosts - 4 addresses
Sat Feb  9 23:40:40 2019 daemon.info dnsmasq[10813]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sat Feb  9 23:40:40 2019 daemon.info dnsmasq-dhcp[10813]: read /etc/ethers - 0 addresses
Sat Feb  9 23:40:40 2019 user.notice omr-bypass: OMR-ByPass is running
Sat Feb  9 23:40:41 2019 daemon.notice netifd: omrvpn (6546): udhcpc: sending select for 10.255.255.28
Sat Feb  9 23:40:41 2019 daemon.notice netifd: omrvpn (6546): udhcpc: lease of 10.255.255.28 obtained, lease time 7200
Sat Feb  9 23:40:41 2019 daemon.notice netifd: Interface 'omrvpn' is now up
Sat Feb  9 23:40:41 2019 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sat Feb  9 23:40:41 2019 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sat Feb  9 23:40:44 2019 user.notice omr-bypass: Starting OMR-ByPass...
Sat Feb  9 23:40:44 2019 daemon.info dnsmasq[10813]: read /etc/hosts - 4 addresses
Sat Feb  9 23:40:44 2019 daemon.info dnsmasq[10813]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sat Feb  9 23:40:44 2019 daemon.info dnsmasq-dhcp[10813]: read /etc/ethers - 0 addresses
Sat Feb  9 23:40:44 2019 user.notice omr-bypass: OMR-ByPass is running
Sat Feb  9 23:40:49 2019 user.notice post-tracking-post-tracking: omrvpn down
Sat Feb  9 23:40:49 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 172.18.51.1 dev eth2.101
Sat Feb  9 23:40:49 2019 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Sat Feb  9 23:40:49 2019 user.notice Shadowsocks: Rules DOWN
Sat Feb  9 23:40:49 2019 user.notice Shadowsocks: Rules UP
Sat Feb  9 23:40:49 2019 user.notice omr-bypass: Starting OMR-ByPass...
Sat Feb  9 23:40:50 2019 daemon.info dnsmasq[10813]: read /etc/hosts - 4 addresses
Sat Feb  9 23:40:50 2019 daemon.info dnsmasq[10813]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sat Feb  9 23:40:50 2019 daemon.info dnsmasq-dhcp[10813]: read /etc/ethers - 0 addresses
Sat Feb  9 23:40:50 2019 user.notice omr-bypass: OMR-ByPass is running
Sat Feb  9 23:41:02 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sat Feb  9 23:41:02 2019 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Sat Feb  9 23:41:02 2019 user.notice Shadowsocks: Rules DOWN
Sat Feb  9 23:41:02 2019 user.notice Shadowsocks: Rules UP
Sat Feb  9 23:41:02 2019 user.notice omr-bypass: Starting OMR-ByPass...
Sat Feb  9 23:41:03 2019 daemon.info dnsmasq[10813]: read /etc/hosts - 4 addresses
Sat Feb  9 23:41:03 2019 daemon.info dnsmasq[10813]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sat Feb  9 23:41:03 2019 daemon.info dnsmasq-dhcp[10813]: read /etc/ethers - 0 addresses
Sat Feb  9 23:41:03 2019 user.notice omr-bypass: OMR-ByPass is running

Ysurac, 2019-02-09:
Strange. The VPN is not able to get the DHCP from the VPS server and at the same time there is some shadowsocks problems... You can update to latest VPS script.

kapook007, 2019-02-10:
ok i test it
i will report you

kapook007, 2019-02-10:
now OMRVPN with  dhcp is ok
OMRVPN  got ip every 1hr. 
No disconnet 

whai is "daemon.err /usr/bin/ss-redir[8278]: remote recv: Host is unreachable"?

Sun Feb 10 19:11:46 2019 daemon.err uhttpd[5730]: luci: accepted login on / for root from 192.168.10.102
Sun Feb 10 19:16:17 2019 daemon.err /usr/bin/ss-local[8276]: getpeername: Socket not connected
Sun Feb 10 19:37:15 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Sun Feb 10 19:37:22 2019 daemon.err omr-tracker[7662]: Segmentation fault
Sun Feb 10 19:37:22 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sun Feb 10 19:38:26 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Sun Feb 10 19:38:37 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sun Feb 10 19:42:01 2019 daemon.err /usr/bin/ss-local[8276]: getpeername: Socket not connected
Sun Feb 10 19:42:04 2019 daemon.err /usr/bin/ss-redir[8277]: remote recv: Host is unreachable
Sun Feb 10 19:43:18 2019 daemon.err /usr/bin/ss-local[8276]: getpeername: Socket not connected
Sun Feb 10 19:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: sending renew to 10.255.255.1
Sun Feb 10 19:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: bind(UDP): Address not available
Sun Feb 10 19:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: sending renew to 0.0.0.0
Sun Feb 10 19:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: lease of 10.255.255.28 obtained, lease time 7200
Sun Feb 10 20:00:00 2019 cron.info crond[2844]: USER root pid 7561 cmd /etc/init.d/vnstat_backup backup
Sun Feb 10 20:00:00 2019 user.info vnstat_backup: backing up database
Sun Feb 10 20:01:46 2019 daemon.err /usr/bin/ss-redir[8278]: remote recv: Host is unreachable
Sun Feb 10 20:04:43 2019 daemon.err /usr/bin/ss-local[8276]: getpeername: Socket not connected
Sun Feb 10 20:37:23 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Sun Feb 10 20:37:37 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sun Feb 10 20:38:44 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Sun Feb 10 20:38:55 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sun Feb 10 20:43:21 2019 daemon.err /usr/bin/ss-redir[8278]: remote recv: Host is unreachable
Sun Feb 10 20:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: sending renew to 10.255.255.1
Sun Feb 10 20:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: bind(UDP): Address not available
Sun Feb 10 20:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: sending renew to 0.0.0.0
Sun Feb 10 20:48:14 2019 daemon.notice netifd: omrvpn (6550): udhcpc: lease of 10.255.255.28 obtained, lease time 7200
Sun Feb 10 20:50:17 2019 daemon.err /usr/bin/ss-redir[8277]: remote recv: Host is unreachable
Sun Feb 10 20:56:20 2019 daemon.err uhttpd[5730]: luci: accepted login on /admin/services/iperf for root from 58.8.187.230

Ysurac, 2019-02-11:
It's what is said: a host can't be reach by the VPS. Not a problem.

closed 2019-02-11 by Ysurac