Ysurac/openmptcprouter #355: Bug Re-create fullmesh subflows

kapook007, 2019-02-11:
Bug Re-create fullmesh subflows
wan with dynamic ip, many isp disconnet every 24hr.
when wan use pppoe disconnet and reconnet in 2-3 second.

OMR-Tracker can not  detect 
no this
 post-tracking-post-tracking: eth1 switched off
 post-tracking: Multipath eth1 switched to on

when wan recive new ip router can not Re-create fullmesh subflows.

can you update OMR-Tracker check present ip wan .
if ip wan change run script 
 post-tracking-post-tracking: eth1 switched off
 post-tracking: Multipath eth1 switched to on

 Re-create fullmesh subflows working.
thanks.

Ysurac, 2019-02-11:
When IP change you should have a "Reloading mtpcp config due to...." message in log. Can you check if you have this?

kapook007, 2019-02-12:
No have "Reloading mtpcp config due to...."
My wan behind nat.

Tue Feb 12 01:42:13 2019 user.notice post-tracking-post-tracking: Multipath eth0.101 switched to on
Tue Feb 12 01:42:13 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Tue Feb 12 01:42:24 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Tue Feb 12 01:42:39 2019 user.notice post-tracking-post-tracking: Set server vps (61.xx.xx.xx) route via 172.18.51.1 metric 5
Tue Feb 12 01:53:07 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 01:54:19 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 01:58:18 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 02:00:00 2019 cron.info crond[3136]: USER root pid 31294 cmd /etc/init.d/vnstat_backup backup
Tue Feb 12 02:00:00 2019 user.info vnstat_backup: backing up database
Tue Feb 12 02:01:02 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 02:03:58 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 02:06:02 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 02:07:35 2019 daemon.err /usr/bin/ss-local[7462]: getpeername: Socket not connected
Tue Feb 12 02:20:19 2019 daemon.err uhttpd[5093]: luci: accepted login on / for root from 124.121.xx.xx

Ysurac, 2019-02-12:
Strange. I will change some settings for next minor release.

closed 2019-07-04 by Ysurac