Ysurac/openmptcprouter #356: Latence et VPN / Latency and VPN

Randdrick, 2019-02-12:
Hello

Firstly a big thank you and congratulations for this solution that I just discovered.

I'm trying to implement OpenMPTCPRouter to increase my flow and stabilize my internet connection. For this I have a connection ADSL Orange (stable at 11 Mbps max) and a 4G to 100 Mbps connection (but with a variation in latency that may at times exceed 100 ms)

After conducting many tests, here are my findings:

-Interface default master seems to be the one who delivers the most debit. 
 I tried to force the default orange connection to see if it allowed me to reduce, stabilize my latency, but systematically my 4G connection will return to master.

-GloryTun is my default VPN (setting to use TCP). I tried to see what it gave in UDP, and I could see a loss of 100% of packages when I am streaming.... This packet loss that I suffered is essentially linked to my 4G connection, it's a fact. What I can't explain, that is why OpenMTCPRouter is not based on the most stable connection (which would avoid too high latency due to excessive loss of package)?
 Is whether a WAN interface master has an impact on the stability of the connection? (Which would imply that my WAN1 being my most stable connection, and becoming a Master, me would lend stability at the expense of a better flow?)

-Last point: as explained above, and due to the instability of my connection, I suffered the loss of packets that led to the latency. However, this latency rarely exceeds 100 ms (without VPN or by setting my LAN in OverOMR interface).

When my vpn is active and if I do a ping on google for example test, my latency remains stable. See even more stable than normal because as a whole it is between 40 and 60 ms. There may be a peak 100 MS see more, but this is rare.

As soon as I do streaming, my latency explodes quickly and can reach easily the 3000 or 4000 ms in advanced (with active VPN) without the VPN, it will rarely exceed 150 ms. Even though I know for a fact that the control mechanism of the packets generated TCP can be the source of the increase of this latency, let me hurt such amplification.
Hoping that my remarks and my questions can make tracks of reflection and improvement to this project which is really good.

==============================
## Specifications

  - OpenMPTCProuter version: (V.0.46.3)
  - OpenMPTCProuter VPS version: (V.093
  - OpenMPTCProuter platform: (VirtualBox x86_64)
==============================

## En français :

Bonjour,
Tout d'abord un grand merci et bravo pour cette solution que je viens de découvrir.
Je suis en train de mettre en place OpenMPTCPRouter afin d'augmenter mon débit et stabiliser ma connexion internet. Pour cela je dispose d'une connexion ADSL Orange (stable à 11 Mbits max) et d'une connexion 4G à 100 Mbits (mais avec une variation de latence pouvant par moment dépasser les 100 ms)
Après avoir mené de nombreux tests, voici mes premières constatations :
- L'interface master par défaut semble être celle qui délivre le plus de débit. 
J'ai bien essayé de forcer la connexion orange par défaut pour savoir si cela me permettait de diminuer, voir stabiliser ma latence, mais systématiquement ma connexion 4G se remet en master.
- GloryTun est mon VPN par défaut (paramétré pour utiliser le protocole TCP). J’ai bien essayé de voir ce que cela donnait en UDP, et j'ai pu constater une perte de 100 % des paquets lorsque je fais du streaming... . Cette perte de paquet que je subis est essentiellement liée à ma connexion 4G, c'est un fait. Ce dont je ne m'explique pas, c'est pourquoi OpenMTCPRouter ne se base pas sur la connexion la plus stable (ce qui éviterait une trop grande latence suite à une perte trop importante de paquet) ?
Est-ce que le fait qu'une interface WAN soit en master a une incidence sur la stabilité de la connexion ? (Ce qui supposerait que mon WAN1 étant ma connexion la plus stable, et en devenant Master, me confèrerait une meilleure stabilité au détriment d'un meilleur débit ?)
- Dernier point : Comme expliqué plus haut, et dû à l'instabilité de ma connexion, je subis de la perte de paquets qui entraine de la latence. Néanmoins, cette latence dépasse rarement les 100 ms (sans VPN ou en paramétrant mon interface LAN en OverOMR).
Lorsque mon vpn est actif et si je fais un test de ping sur google par exemple, ma latence reste stable. Voir même plus stable qu'en temps normal puisqu'en globalité elle se situe entre 40 et 60 ms. Il peut y avoir une pointe à 100 ms voir plus, mais c'est rare.
Dès que je fais du streaming, ma latence explose très vite et peut atteindre facilement les 3000 ou 4000 ms en pointe (avec le VPN actif) Sans le VPN, elle dépassera rarement les 150 ms. Même si je sais pertinemment que le mécanisme de contrôle des paquets qu'engendre TCP puisse être la source de l'augmentation de cette latence, je m'explique mal une telle amplification.
En espérant que mes remarques et mes questions puissent apporter des pistes de réflexion et d'amélioration à ce projet qui est vraiment très bien.

Ysurac, 2019-02-12:
* In the interface, OpenMPTCProuter menu, advanced parameters, you can set how the master interface is selected in "Master interface selection". By default the interface with lower RTT should be used as master. This seems to not work in your case, I need the result of ```uci show openmptcprouter``` via SSH.
* So you tester Glorytun TCP and Glorytun UDP ? And it's not working at all with Glorytun UDP ? And master set on WAN1 should give a stable connection.
* Streaming use UDP ? I know there is a problem with UDP... You can try to change the congestion control and see if it's better.

Randdrick, 2019-02-12:
Here is the result of the command. I deliberately truncated some information like the IP address of my VPS, the password and its token.

root@OpenMPTCProuter:~#  uci show openmptcprouter
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.shadowsocks_disable='0'
openmptcprouter.settings.vpn='glorytun_tcp'
openmptcprouter.settings.master_lcintf='wan1' 'wan2'
openmptcprouter.settings.version='0.46.3'
openmptcprouter.settings.master='dynamic'
openmptcprouter.vps=server
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.ip='188.213.'
openmptcprouter.vps.password=''
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.available_vpn='glorytun-tcp' 'glorytun-udp' 'openvpn' 'mlvpn'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.kernel='4.14.94-mptcp-a284ba1'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.93'
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.token=''
openmptcprouter.omr=router
openmptcprouter.omr.shadowsocks='up'
openmptcprouter.omr.detected_public_ipv4='188.213.'
openmptcprouter.omr.detected_ss_ipv4='188.213.'
openmptcprouter.wan1=interface
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.asn='AS3215'
openmptcprouter.wan1.mptcp_status='-tracebox'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.publicip='2.6.86.29'
openmptcprouter.wan1.lc='1549997116'
openmptcprouter.wan1.latency_previous='34'
openmptcprouter.wan1.latency='32'
openmptcprouter.wan2=interface
openmptcprouter.wan2.mtu='1500'
openmptcprouter.wan2.asn='EI-TELECOM'
openmptcprouter.wan2.mptcp_status='-tracebox'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.publicip='193.56.242.111'
openmptcprouter.wan2.lc='1549997131'
openmptcprouter.wan2.latency_previous='56'
openmptcprouter.wan2.latency='123'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.mtu='1500'
openmptcprouter.omrvpn.state='down'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.omr='0.46.3'
openmptcprouter.latest_versions.vps='0.93'
openmptcprouter.latest_versions.lc='1549997115'

- I just forced the WAN1 interface into master and change the mode in the advanced settings, to put it on dynamic. For the moment, the WAN1 interface remains in master.

- I disabled the OMRVPN interface in the OMR-Tracker menu.
I'm observing less latency with GloryTun TCP. However, this latency is always more important than it should be since it can mount up to 700 MS (about) while it is up to 150 ms on my interface WAN2 

- I tested Glorytun TCP and Glorytun UDP. I can confirm that my WAN1 connection is very stable. With Glorytun UDP, I lost 100% of the packets in streaming. 

- Streaming can also use UDP to make a connection more fluid. When this is the case, and if I use only my 4g connection, I observe a loss of packets of 30%, in bad conditions. Under correct conditions, I do not exceed 10% loss. 

- I have done other tests and I think I have found the cause of the problem. If I ask to do streaming by focusing on UDP, I can see a loss of packet (even with GloryTun TCP). And therefore a very strong increase in my latency.
Conversely, if I prefer streaming to TCP, I no longer have a significant increase in my latency. So I infer that GloryTun really has a problem with UDP.

HenryJacques, 2019-05-27:
Hello

>     * Streaming use UDP ? I know there is a problem with UDP... You can try to change the congestion control and see if it's better.

Can you elaborate on this? (handling UDP traffic)
My use case is to share a UDP traffic between several links. Are you saying OMR can't do that at that time?

Thanks

closed 2019-07-04 by Ysurac