Ysurac/openmptcprouter #357: Suggestion: Bridges for LAN, WAN1, WAN2, etc

lars18th, 2019-02-13:
Hi,

By default, OpenWRT instead of use "eth0" as the default LAN interface, it uses "br-lan" and bridges the "eth0" to this bridge. This simplifies the configuration of the device when using multiple interfaces for LAN (for example, with wifi and ethernet). But it resolves also the case of special environments. For example, when using L2 tunnels in the LAN.

Summarizing: Instead of use a physical interface, the default config uses a bridge for LAN. So the configuration is agnostic about the interfaces.

I suggest to implement this in OMR, but not only for LAN (br-lan), but for WAN1 (br-wan1), WAN2 (br-wan2), etc.

With this, will be possible to create complex topologies, like (as an example) a device with just one Ethernet port, and 3xWAN+1xLAN without the use of VLANs (using then L2 tunnels).

The degradation of performance with this change will be minimal. And if you care about it, then support both options: with and without bridges.

You agree?

PhatHub, 2019-04-16:
I think being able to play with bridges would be good for complexity. Example: There's WANs that have the same IP address, like multiple Android Wifi tethering (192.168.43.1) or multiple Android USB tethering (192.168.42.1) could be bridged into different IP address subnets to de-conflict .

lars18th, 2019-04-16:
Yes. For _several_ scenarios support for Bridges in OMR is good!

Ysurac, 2019-04-16:
You can already use bridges if you want. It's only not default configuration.

lars18th, 2019-04-16:
> You can already use bridges if you want. It's only not default configuration.

Some months ago I do some tests and I can't do it!
WAN1 and WAN2 are "interfaces" and not "bridges".
My suggestion is to move the default "WAN1" to "br-WAN1" and manage it every time as it in OMR.
If only one interface (current WAN1) is in the bridge nothing changes to you.
But we can "bridge" it to any other interface without troubles.

It's just with "br-lan" by default.

I hope you like to consider it.

closed 2019-07-04 by Ysurac

lars18th, 2019-07-05:
Hi @Ysurac ,

I hope you will reconsider your position about WANs over BRIDGES! That's the reasons that I see:
- Without them is not possible to use PPPoE connections as the WAN interfaces.
- If the user requires to move from "WAN1" to "br-WAN1" it's very hard to do it, as you assume that all WAN connections are DEVICES.

Please, don't think that this adds complexity: The default "br-lan" uses the same concept, even if the OMR uses only one interface for the LAN.

Regards.

Ysurac, 2019-07-05:
For now I don't have the time to check how to do that, if you submit a working configuration I can test and add it :)
I want to release a working and good 4.19 version, then I will check suggestions like that.
So it's a close "for now".

lars18th, 2019-07-05:
> For now I don't have the time to check how to do that, if you submit a working configuration I can test and add it :)

Thank you!

> I want to release a working and good 4.19 version, then I will check suggestions like that.
> So it's a close "for now".

I'm glad to hear it... I see you're a reasonable person.

Now I'm very busy, but I'll try to define a working configuration with a repoducible environment.
We'll keep in touch! :+1: