Ysurac/openmptcprouter #359: can't get public IP/connect to VPS when connections are all USB LTE

dnwk, 2019-02-14:
Here is System Log
Thu Feb 14 04:30:23 2019 daemon.err uhttpd[6245]: Failed to establish connection
Thu Feb 14 04:31:24 2019 daemon.err uhttpd[6245]: Failed to establish connection
Thu Feb 14 04:32:04 2019 daemon.err uhttpd[6245]: Failed to establish connection
Thu Feb 14 04:32:50 2019 daemon.err uhttpd[6245]: Failed to establish connection
Thu Feb 14 04:33:31 2019 daemon.err uhttpd[6245]: Failed to establish connection

See below screenshot. At the same time, SS is actually running fine without a problem. 

[image]

## Specifications

  - OpenMPTCProuter version: (Last is not a version) v0.46.3 r0
  - OpenMPTCProuter VPS version: (Last is not a version) 0.93
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) Linksys WRT32X

dnwk, 2019-02-14:
Additional Info: In Interface, USB LTE did not receive an WAN name. It's called 3g-LTE. I can't figure out how to get it a LTE name. So that in OMR-Bypass, I can't set to traffic to bypass into 3g-LTE as it is not in the drop down.

dnwk, 2019-02-14:
[image]

dnwk, 2019-02-14:
My 3g-LTE is on /dev/ttyACM0

Ysurac, 2019-02-14:
Hi, I would need the result of ```uci show``` via SSH on OpenMPTCProuter (don't forget to remove server IP and key).

dnwk, 2019-02-15:
firewall.@rule[17].icmp_type='echo-request'
firewall.@redirect[2]=redirect
firewall.@redirect[2].target='DNAT'
firewall.@redirect[2].src='vpn'
firewall.@redirect[2].dest='lan'
firewall.@redirect[2].proto='tcp'
firewall.@redirect[2].src_dport='8443'
firewall.@redirect[2].dest_ip='192.168.100.175'
firewall.@redirect[2].dest_port='8443'
firewall.@redirect[2].name='8443'
firewall.@redirect[3]=redirect
firewall.@redirect[3].target='DNAT'
firewall.@redirect[3].src='vpn'
firewall.@redirect[3].dest='lan'
firewall.@redirect[3].proto='udp'
firewall.@redirect[3].src_dport='9201'
firewall.@redirect[3].dest_ip='192.168.100.175'
firewall.@redirect[3].dest_port='9201'
firewall.@redirect[3].name='9201'
firewall.@redirect[4]=redirect
firewall.@redirect[4].target='DNAT'
firewall.@redirect[4].src='vpn'
firewall.@redirect[4].dest='lan'
firewall.@redirect[4].proto='tcp'
firewall.@redirect[4].src_dport='32400'
firewall.@redirect[4].dest_ip='192.168.100.175'
firewall.@redirect[4].dest_port='32400'
firewall.@redirect[4].name='32400'
firewall.@redirect[5]=redirect
firewall.@redirect[5].target='DNAT'
firewall.@redirect[5].src='vpn'
firewall.@redirect[5].dest='lan'
firewall.@redirect[5].proto='tcp'
firewall.@redirect[5].src_dport='44330'
firewall.@redirect[5].dest_ip='192.168.100.175'
firewall.@redirect[5].dest_port='44330'
firewall.@redirect[5].name='44330'
firewall.@redirect[6]=redirect
firewall.@redirect[6].target='DNAT'
firewall.@redirect[6].src='vpn'
firewall.@redirect[6].dest='lan'
firewall.@redirect[6].proto='tcp'
firewall.@redirect[6].src_dport='8080'
firewall.@redirect[6].dest_ip='192.168.100.125'
firewall.@redirect[6].dest_port='8080'
firewall.@redirect[6].name='piaware'
firewall.@redirect[7]=redirect
firewall.@redirect[7].target='DNAT'
firewall.@redirect[7].src='vpn'
firewall.@redirect[7].dest='lan'
firewall.@redirect[7].proto='tcp'
firewall.@redirect[7].src_dport='443'
firewall.@redirect[7].dest_ip='192.168.100.175'
firewall.@redirect[7].dest_port='443'
firewall.@redirect[7].name='HTTPS'
firewall.@rule[18]=rule
firewall.@rule[18].enabled='1'
firewall.@rule[18].target='ACCEPT'
firewall.@rule[18].name='Allow-All-Ping'
firewall.@rule[18].proto='icmp'
firewall.@rule[18].dest='*'
firewall.@rule[18].src='*'
firewall.@rule[18].icmp_type='echo-request'
firewall.zone_vpn=zone
firewall.zone_vpn.name='vpn'
firewall.zone_vpn.masq='1'
firewall.zone_vpn.input='REJECT'
firewall.zone_vpn.forward='ACCEPT'
firewall.zone_vpn.output='ACCEPT'
firewall.zone_vpn.network='glorytun omrvpn omr6in4'
[... 933 lines cut ...]
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.hwmode='11g'
wireless.radio1.path='soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
wireless.radio1.country='US'
wireless.radio1.legacy_rates='1'
wireless.radio1.channel='auto'
wireless.radio1.htmode='HT40'
wireless.radio2=wifi-device
wireless.radio2.type='mac80211'
wireless.radio2.path='platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
wireless.radio2.country='US'
wireless.radio2.legacy_rates='1'
wireless.radio2.hwmode='11g'
wireless.radio2.channel='auto'
wireless.radio2.htmode='HT40'
wireless.radio2.txpower='23'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].device='radio2'
wireless.@wifi-iface[0].mode='ap'
wireless.@wifi-iface[0].network='lan'
wireless.@wifi-iface[0].ssid='*'
wireless.@wifi-iface[0].disassoc_low_ack='0'
wireless.@wifi-iface[0].encryption='psk-mixed'
wireless.@wifi-iface[0].key='*'

Ysurac, 2019-02-15:
One interface must be set as multipath master.
Else I have no idea, I don't have a 3G key.

dnwk, 2019-02-16:
Yes. I have set it as master. It doesn't seem to improve. And maybe it is a upstream problem on OpenWRT side. Because I can't ping anything from the router. However, internet access/shadowsocks/glorytun works fine.

dnwk, 2019-03-28:
This seem to work in x86 version. I test it on a x86 with USB LTE modem only and seems fine.

closed 2019-03-28 by dnwk