Ysurac/openmptcprouter #361: VPN Tunnel Down, bad config ?

victorleduc, 2019-02-14:
## Expected Behavior

  VPN should be UP

## Actual Behavior

Hi, I think I have a config issue, but can't really tell what I did wrong.
Here are some screenshots : 
[image]
[image]
[image]
[image]

I can't find the OpenVPN key in the VPS, where it is, and it is why it's not working ?

Thanks for the help :)

## Specifications

  - OpenMPTCProuter version: v0.46.3
  - OpenMPTCProuter VPS version: 0.93
  - OpenMPTCProuter platform: Linksys WRT3200ACM

victorleduc, 2019-02-14:
And here is the logs when I hit "Save and Apply" in the Wizard :
[image]

victorleduc, 2019-02-14:
And here is the result of uci show
[uci_show.txt](https://github.com/Ysurac/openmptcprouter/files/2865045/uci_show.txt)

zicochaos, 2019-02-14:
disable ipv6

victorleduc, 2019-02-14:
Hi, if I disable ipv6, it only make the orange message "VPN Ipv6 tunnel down", but I still got "VPN tunnel down"

zicochaos, 2019-02-14:
setup address for at least two wan's

victorleduc, 2019-02-14:
It seems strange that it can not work with only one WAN, I wanted to config it waiting for my VDSL line in a few weeks. And I'm not sure but I think I saw screenshot of working VPN with only one WAN.
Can you confirm @yrusac ?

victorleduc, 2019-02-14:
After searching a little, i saw that my server always had SSH port 22 and not 65222 (and supposedly Shadowsocks/Glorytun too?), so I needed to reboot the VPS and it is now better : 
[image]
Can you help me with the last error ?

victorleduc, 2019-02-14:
I just waited a little and apparently it started working ! 😄
[image]

Ysurac, 2019-02-14:
I'm happy to help to resolve this issue by doing nothing :smile: 
If you don't use them you should disable multipath on wan2 to wan4 interface in Network->MPTCP interface.

victorleduc, 2019-02-14:
Done :)

closed 2019-02-14 by victorleduc