Ysurac/openmptcprouter #363: Problem when restoring backup

MrDaweb, 2019-02-15:
The parameters of the "Smart Queue Management" part are not well taken over: The two entries take the value Eth0 instead of Wan1 and Wan2 (which caused a sharp decrease in flow)

## Steps to Reproduce the Problem

  1. Restore backup on new installation

## Specifications

  - OpenMPTCProuter version: (v0.46.3 r0+9228-a3ccac6b1d / LuCI Master (git-19.037.70147-7beebee))
  - OpenMPTCProuter VPS version: (0.93)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-02-15:
Backup from same version or a previous version ? If previous version from which version ?

MrDaweb, 2019-02-15:
It's from 0.44

cron410, 2019-03-14:
@MrDaweb where do you find the setting for "Smart Queue Management"? I assume this is for QOS purposes?

Ysurac, 2019-03-14:
In network SQM and in OpenMPTCProuter wizard.

closed 2019-07-04 by Ysurac