Ysurac/openmptcprouter #364: Netflix ORM

MrDaweb, 2019-02-16:
## Expected Behavior

## Actual Behavior

The NETFLIX traffic still does not work despite the ORM declaration (wan 1 or 2). I also tried ASN2906 on the Wan2, always the same. Netflix tells me about a proxy or whatever, and I have to manually switch to one of my connections for it to work.

## Specifications

  - OpenMPTCProuter version: v0.46.3 r0+9228-a3ccac6b1d / LuCI Master (git-19.037.70147-7beebee)
  - OpenMPTCProuter VPS version: (0.93)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-02-16:
You have to use OMR-ByPass. This can take a few minutes before really working if data are on client cache.

MrDaweb, 2019-02-16:
Yes of course, that's what I say, I activated ORM-Bypass on Wan2, (and I also tried Wan 1) and it never worked. I updated from 0.44 to 0.46 for two days, but that did not change anything. emptied the cache, restarted the router and the VPS (at OVH), nothing helps!

I have another installation ditto in another office ditto, with another VPS at OVH and it works!

Ysurac, 2019-02-16:
Do you use OpenMPTCProuter as DNS ?

MrDaweb, 2019-02-19:
Well seen ! I had defined on the DHCP of the OpenMPTCProuter dns in manual (8.8.8.8 and 9.9.9.9).
By cons blow or I tell the OTB to use selected DNS? On the Lan Interface? On the Wan?

Ysurac, 2019-02-19:
It's not a good idea to use public DNS with OpenMPTCProuter. By default root DNS server are used.
So OpenMPTCProuter IP address must be used on client, and no DNS set for each interfaces defined in OpenMPTCProuter.

MrDaweb, 2019-02-20:
Perfect, I thought he was using the basic DNS operators. So I do not touch anything and I stay like that, it works perfectly;)

closed 2019-02-20 by MrDaweb