Ysurac/openmptcprouter #366: VPN tunnel DOWN

Peppe2201, 2019-02-17:
Bonding of three WAN don't work, only one WAN is used, the Status Page result VPN tunnel DOWN on the router icon.
In my config i've added a second LAN with DHCP server but i don't think is that the problem, i've the same problem even deleting that interface
[image]

## Steps to Reproduce the Problem

  1. Configure the router with the wizard adding the 3 wan in DHCP mode
  2. Add in the wizard the IP and all the key for the VPN service retrived on the VPS

## Specifications

  - OpenMPTCProuter version: 0.44.5
  - OpenMPTCProuter VPS version: 0.91
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-02-17:
You should update.

Yanaem, 2019-02-21:
I have the same problem and i am updated. 

OpenMPTCProuter v0.46
OpenMPTCProuter VPS v0.90

Could you help me?

[image]

Yanaem, 2019-02-21:
Ok now it's good i don't know why...

Ysurac, 2019-02-22:
You should check the system log. The VPN can be down when there is a master connection problem.

kapook007, 2019-02-26:
i have same problem VPN tunnel DOWN
when run system over 5-10hr  , restart openmptcprouter , glorytun tcp can not recive ip  from vps .
i restat openmptcprouter  not recive ip .
i restat VPS  my system working openmptcprouter  can recive ip .

my log

Tue Feb 26 12:05:51 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Feb 26 12:05:51 2019 user.notice SQM: Stopping SQM on eth0.201
Tue Feb 26 12:05:51 2019 daemon.info glorytun[10529]: INITIALIZED tun0
Tue Feb 26 12:05:51 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Tue Feb 26 12:05:51 2019 daemon.notice netifd: Network device 'tun0' link is up
Tue Feb 26 12:05:51 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Tue Feb 26 12:05:51 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Tue Feb 26 12:05:51 2019 daemon.notice netifd: omrvpn (10696): udhcpc: started, v1.30.0
Tue Feb 26 12:05:51 2019 daemon.notice netifd: omrvpn (10696): udhcpc: sending discover
Tue Feb 26 12:05:51 2019 user.notice SQM: Starting SQM script: simple.qos on eth0.201, in: 50000 Kbps, out: 50000 Kbps
Tue Feb 26 12:05:54 2019 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.206)
Tue Feb 26 12:05:54 2019 daemon.notice netifd: omrvpn (10696): udhcpc: sending discover
Tue Feb 26 12:05:54 2019 user.notice mptcp: Reloading mptcp config due to ifup of wan6 (eth0.206)
Tue Feb 26 12:05:55 2019 user.notice mptcp: Reloading mptcp config due to ifup of omr6in4 (6in4-omr6in4)
Tue Feb 26 12:05:56 2019 daemon.err omr-tracker-ss[11652]: Failed to establish connection
Tue Feb 26 12:05:57 2019 daemon.notice netifd: omrvpn (10696): udhcpc: sending discover
Tue Feb 26 12:05:57 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Tue Feb 26 12:05:58 2019 user.notice post-tracking-post-tracking: Set server vps (122.155.xx.xx) route via 172.18.52.1 metric 6
Tue Feb 26 12:06:00 2019 user.notice post-tracking-post-tracking: omrvpn down
Tue Feb 26 12:06:00 2019 user.notice post-tracking-post-tracking: Set server vps (122.155.xx.xx) route via 172.18.56.1 metric 10
Tue Feb 26 12:06:00 2019 user.notice post-tracking-post-tracking: Set server vps (122.155.xx.xx) route via 172.18.54.1 metric 8
Tue Feb 26 12:06:00 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 172.18.51.1 dev eth0.201
Tue Feb 26 12:06:00 2019 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Tue Feb 26 12:06:00 2019 user.notice Shadowsocks: Rules DOWN
Tue Feb 26 12:06:00 2019 user.notice Shadowsocks: Rules UP

cron410, 2019-03-14:
I have the same issue, always VPN Tunnel DOWN. 

```
Thu Mar 14 02:45:52 2019 user.notice post-tracking-post-tracking: omrvpn down
Thu Mar 14 02:45:52 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.1.1 dev wan2
Thu Mar 14 02:45:52 2019 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Thu Mar 14 02:45:53 2019 user.notice Shadowsocks: Rules DOWN
Thu Mar 14 02:45:53 2019 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.0.1 dev wan3
Thu Mar 14 02:45:53 2019 user.notice Shadowsocks: Rules UP
Thu Mar 14 02:46:35 2019 daemon.info dnsmasq[8085]: read /etc/hosts - 4 addresses
Thu Mar 14 02:46:35 2019 daemon.info dnsmasq[8085]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Thu Mar 14 02:46:35 2019 user.notice Shadowsocks: Rules DOWN
Thu Mar 14 02:46:35 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Thu Mar 14 02:46:36 2019 user.notice Shadowsocks: Rules UP
Thu Mar 14 02:46:36 2019 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Thu Mar 14 02:46:36 2019 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Thu Mar 14 02:46:36 2019 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: using tcp fast open
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: enable TCP no-delay
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: initializing ciphers... chacha20
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: listening at 127.0.0.1:1111
Thu Mar 14 02:46:36 2019 user.info : using tcp fast open
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: enable TCP no-delay
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: resolving hostname to IPv6 address first
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: initializing ciphers... chacha20
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: listening at 0.0.0.0:1100
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: tcp port reuse enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: UDP relay enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: udp port reuse enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19435]: running from root user
Thu Mar 14 02:46:36 2019 user.info : using tcp fast open
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: enable TCP no-delay
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: resolving hostname to IPv6 address first
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: initializing ciphers... chacha20
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: listening at 0.0.0.0:1100
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: tcp port reuse enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: UDP relay enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: udp port reuse enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-redir[19434]: running from root user
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: tcp port reuse enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: udprelay enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: udp port reuse enabled
Thu Mar 14 02:46:36 2019 daemon.info /usr/bin/ss-local[19433]: running from root user
Thu Mar 14 02:46:36 2019 user.notice Shadowsocks: Rules DOWN
Thu Mar 14 02:46:37 2019 user.notice Shadowsocks: Rules UP
Thu Mar 14 02:46:37 2019 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Thu Mar 14 02:46:37 2019 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Thu Mar 14 02:46:37 2019 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Thu Mar 14 02:46:37 2019 user.info : using tcp fast open
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: enable TCP no-delay
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: resolving hostname to IPv6 address first
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: initializing ciphers... chacha20
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: listening at 0.0.0.0:1100
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: tcp port reuse enabled
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: UDP relay enabled
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: udp port reuse enabled
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19738]: running from root user
Thu Mar 14 02:46:37 2019 user.info : using tcp fast open
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: enable TCP no-delay
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: resolving hostname to IPv6 address first
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: initializing ciphers... chacha20
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: listening at 0.0.0.0:1100
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: tcp port reuse enabled
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: UDP relay enabled
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: udp port reuse enabled
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-redir[19737]: running from root user
Thu Mar 14 02:46:37 2019 daemon.info /usr/bin/ss-local[19736]: using tcp fast open
[... cut ...]
Thu Mar 14 02:46:44 2019 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.0.1 dev wan3
Thu Mar 14 02:47:11 2019 user.notice post-tracking-post-tracking: Get status and settings for wan3...
Thu Mar 14 02:47:21 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Thu Mar 14 02:47:24 2019 user.notice post-tracking-post-tracking: Get status and settings for wan3... Done
Thu Mar 14 02:47:35 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
```

Ysurac, 2019-03-14:
I don't see any log about glorytun. What VPN are you using ?
I need the result of ```uci show``` via SSH or in System->OpenMPTCProuter->show all settings, remove at least server IP.

cron410, 2019-03-15:
I'm using GloryTun in UDP mode.

```
root@OpenMPTCProuter:~# uci show
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '8.8.8.8'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.ra_management='1'
dhcp.lan.ignore='1'
dhcp.lan.ra_default='0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].network='lan LAN2'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@zone[1].network='wan wan6 wan2 wan3 wan4 wan2 wan3'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
[... 977 lines cut ...]
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'wan2' 'wan3'
```

Peppe2201, 2019-03-19:
I'm still having the same problem, i've update the VPS at the version 0.97 and the router at the version 0.47.1, same problem with the VPN tunnel DOWN and i've noticed than while doing a speedtest the CPU usage is really high (the router have an i5 4 core processor) and the speedtest results really bad performance, only large download use all the bandwith. Other problem i've noticed is that my upload goes only at max 10mbit/s while i've a 100mbit upload line

Peppe2201, 2019-03-19:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.lan.dhcpv6='disabled'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
[... 913 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'

Ysurac, 2019-03-19:
But I still don't know what VPN you are using.
Glorytun UDP seems to not working for now.

Ysurac, 2019-03-19:
OK, I see the problem : openmptcprouter.vps.kernel='4.15.18-10-pve' This can't work with a VPS kernel that doesn't have MPTCP enabled.

Peppe2201, 2019-03-24:
Now the VPS is KVM and I don't have the VPN tunnel down problem but seems that is using only one WAN, i have connected 3 WAN, omr-test-speed is using also only one connection.

[image]

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
[... 887 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3

cron410, 2019-03-26:
I have this same problem if I change GloryTUN to use UDP instead of the default TCP. Even changing it back to TCP does not solve the problem right away, only 1 WAN connection is used in the clear, not through my VPS. After rebooting the local router with GloryTUN set to TCP, everything works correctly. Port Forwards work, aggregated bandwidth works, traceroute works

```
Tracing route to google-public-dns-a.google.com [8.8.8.8]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.100.1
  2    34 ms    53 ms    58 ms  10.255.255.1
  3    59 ms    38 ms    65 ms  xx.xx.xx.xx
  4    37 ms    36 ms    33 ms  te0-0-0-0.agr22.dca01.atlas.co
.31.57]
  5    97 ms    57 ms    71 ms  be3309.ccr41.dca01.atlas.cogen
.173]
  6    59 ms    49 ms    59 ms  be2658.ccr22.iad02.atlas.cogen
.138]
  7    83 ms    66 ms    92 ms  tata.iad02.atlas.cogentco.com
```

closed 2019-07-04 by Ysurac

rifaterdemsahin, 2020-07-26:
I moved to glorytun as default now it started working
[image]