Ysurac/openmptcprouter #367: Port Forwarding

NovaDavid, 2019-02-18:
## Expected Behavior

  Port forwarding doesn't work

## Actual Behavior

When i try to forward some ports like 8000 or 3389, it doesn't work. I try with to different windows server and two differents gateways (4G and fiber).

I followed documentation for port forwarding and SNAT. Firewall is off on my windows server and access work on local network.

## Steps to Reproduce the Problem

  1. Configure port forwarding and SNAT
  2. Test it with OpenMPTCProuter Server public IP
  3.

## Specifications

  - OpenMPTCProuter version: (0.46.3)
  - OpenMPTCProuter VPS version: (0.93)
  - OpenMPTCProuter platform: (/RPI3/x86_64)

Ysurac, 2019-02-18:
DNAT line at the end of /etc/shorewall/rules on the VPS is not commented ?

NovaDavid, 2019-02-19:
Yes there is a '#' but when i remove it, it reappear after a reboot of the VPS.

NovaDavid, 2019-02-19:
It was because in VPS Settings "Redirect all ports" was unchecked.

[image]

But now i have "Can't get public IP address from ShadowSocks" on VPS

Ysurac, 2019-02-19:
I would need the result of ```uci show``` on OpenMPTCProuter via SSH (remove at least server ip before pasting)

NovaDavid, 2019-02-20:
After a reboot of the raspberry and the VPS everything working fine now. Thanks for your help on your work !

closed 2019-02-20 by NovaDavid