Ysurac/openmptcprouter #368: PPPoE WAN connections: Network interface duplicated

lars18th, 2019-02-20:
- OpenMPTCProuter version: openmptcprouter v0.46.3
  - OpenMPTCProuter VPS version: NA
  - OpenMPTCProuter platform: x86_64

I configure "wan1" and "wan2" as "PPPoE" (that's required in my environment; please don't ask about it!).
And in SYSTEM/OpenMPTCProuter/Status" I see:

- wan1 (192.168.10.1) / ip address: 192.168.10.10; mtu: 1450; multipath: on
- wan2 (192.168.20.1) / **Network interface duplicated** ip address: 192.168.20.10; mtu: 1450; multipath: on

But, for sure both connections are different. I feel the problem can be because the PPPoE interface is using the same _physical_ device. Calling to `ifstatus wan1` and `ifstatus wan2` the result for `"device"` is identical. However, this config is valid.

Why this error?
I feel it's a bug.

Ysurac, 2019-02-20:
Network interface are really duplicated. This doesn't work well with MPTCP.
It's why you should use macvlan for MPTCP.

closed 2019-02-20 by Ysurac

lars18th, 2019-02-21:
Hi @Ysurac ,

> Network interface are really duplicated. This doesn't work well with MPTCP.
It's why you should use macvlan for MPTCP.

I'm using macvlan too, but other pourposes! :wink: 
In any case I don't understand why PPPoE connections are conflictive. I'm using them now. But I'll try to manage them on top of macvlan devices.

Perhaps, this the reason for my troubles described at #373. It's possible that the queues are assigned to the "device" instead of the "interface" and for this reason they bound to both.

When I'll have a stable system using PPPoE then I'll describe here how to do it for refence to other users.
Regards.

PD: Please, reopen this issue. I'll post here all details regarding PPPoE connections. Thank you!

lars18th, 2019-02-21:
Regarding how to improve the support of PPPoE WAN connections, read my post: https://github.com/Ysurac/openmptcprouter/issues/373#issuecomment-466022310

lars18th, 2019-02-21:
Hi,

This PR partially solves the problem when you create your wan connections over PPPoE and share the same physical device: https://github.com/Ysurac/openmptcprouter-feeds/pull/4

Regards.

Ysurac, 2019-02-21:
This is always a problem, routing table for MPTCP can't be correct.

lars18th, 2019-02-21:
Hi,

Regarding the HOW TO of "PPPoE connections" (please, re-open the issue):

- When you use PPPoE and create some "macvlan" interface for them, then configure the "macvlan" is "Multipath=off". This is a requirement to not list this interface in the OMR-Status-Network overview.

And...

> This is always a problem, routing table for MPTCP can't be correct.

No! This is completly false. Multipath works as Layer 3, not Layer 2. So, no interference at all when you use the same _DEVICE_ for multiple _INTERFACES_. I noted that some parts of the project comes from the OVH project. I feel the original authors don't like to support multiple interfaces over the same device to simplify, and not by requirement. When _some_ interface doesn't have any IP address (Layer 3) then it can't generate any problem with routing tables.

Think on this: PPPoE has IP, configured over a DEVICE. But the INTERFACE of the DEVICE already exists but without any IP. Where is the problem?
The problem is with the code that detects two interfaces over the same device.
So, please, see my last commit of my PR that solves the problem: https://github.com/Ysurac/openmptcprouter-feeds/pull/4

I'm now using it without troubles!

Regards.

Ysurac, 2019-02-21:
The problem is not MPTCP, but the routing table needed by MPTCP. By using the same interface for multiple network, there is no way to tell where the packets must go. It's why macvlan is used, even if other way are possible.
What is the howto you are speaking about ?

I think I didn't exactly understand what is your configuration, can you send me the result of ```uci show network``` ?