Ysurac/openmptcprouter #369: Request for LAN Gateway

lars18th, 2019-02-20:
- OpenMPTCProuter version: openmptcprouter v0.46.3
  - OpenMPTCProuter VPS version: NA
  - OpenMPTCProuter platform: x86_64

Hi @Ysurac ,
I like to setup the OMR for connecting to a _private_ network. So, my configuration is in overview:

- WAN1 one VPN connection using path-1.
- WAN2 secondaty VPN connection using path-2.
- LAN has connection to my LAN _and_ to Internet.

In this scenario I configured static routes to use the WAN1/WAN2 to connect to the private network. However, I have these problems:

1. After the setup and in each reboot, the Default Gateway (in the LAN connection) is removed from the main table. I suggest to add one configuration option to not remove it.
2. At time the WAN1 is market as "mptcp master" and WAN2 as "mptcp on". When doing this the default gateway is assigned to WAN1. Howerver, when the WAN1 goes down this generates a trouble, as all traffic is routed via this connection. I suggest to configure ALL WAN as "mptcp on" (not "master") nad set all as default gateway using the METRIC. So, when one with lower priority fails the next is used. My suggestion: METRIC 0 for System Gateway (in my case LAN, but the one configured); METRIC 1 for WAN1; METRIC 2 for WAN2; etc.
3. In SYSTEM/OpenMPTCProuter/Status the "Direct output" fails as the VPS server is inside a private network. The IP shown in Network overview correspond to the wan gateway of the LAN connection. I suggest to add one option for "enable/disable direct output".

I hope you can support this scenario (VPS in private networks over WAN+ & Internet over LAN).
Regards.

lars18th, 2019-02-21:
Hi @Ysurac ,

I'm searching for a fix of this... Please, can you tell me where is the code in OMR that removes the system default gateway?

lars18th, 2019-02-21:
Hi @Ysurac ,

Regarding this issue I noted that in some scripts, like `/etc/init.d/mptcp` and `/usr/share/omr/post-tracking.d/post-tracking` you modify a lot the route tables. Futhermore, I see the uci entry `openmptcprouter.settings.uci_route`, but in my install I don't see this entry (`uci show | grep openmptcprouter.settings.uci_route` returns nothing).

So, as I undertand you're doing some complex routing. However, I suggest to add some support:
1. Custom static routes: At time I feel you remove _all_ static routes from the config. However, I need to add some "user static routes" to my OMR. Please, don't remove routes that are not default.
2. Configurable gateway: I suspect you remove _all_ default gateways in the OMR instance to route all traffic over the tunnel to the VPS. OK. However, then you can't handle the case when all wan routes fail. Even if the LAN network can provide a default gateway. So, I suggest to support one customizable "USER CONFIGURED GATEWAY" with a desired Metric. Then, the user can use, for example, the Metric 250 for the LAN gateway as a last result.

Please, consider to support this.
Regards.

Ysurac, 2019-02-21:
openmptcprouter.settings.uci_route is not defined, this is to define route using OpenWrt route system.
1. I don't remove static routes if it's not conflicting with MPTCP routes.
2. When all wan routes fails, then nothing is working, it's a router... Lan as gateway is a strange solution...

closed 2019-02-21 by Ysurac

lars18th, 2019-05-01:
> 2. When all wan routes fails, then nothing is working, it's a router... Lan as gateway is a strange solution...

That's your opinion. From my point of view MPTCP router has sense when it's **NOT** the only gateway in the LAN. For instance, I prefer (and I use it with) this scenario:

```

                              -------- Gateway 1 ------> Internet over ISP A
                             /
local devices  <-----LAN-----+-------- Gateway 2 ------> Internet over ISP B
                             |
                             +-- MPTCP (default GW) -- > (GW1+GW2 bonding) <----> VPS MPTCP <----> Internet
                             \
                              ---- 4G Caped Modem -----> Internet over UMTS
```

And it's relevant to comment that in this scenario:
- It have sense to **NOT** configure and **NOT** mange the 4G Modem connection.
- This 4G Connection can be any other connection, for example public WiFi o whatever.

Then the **ONLY** solution to have a _secondary_ not managed GW inside the LAN, is that the MPTCP Router can configure the default gw over the LAN. As it's impossible to configure a WAN over the same LAN network.

You get it?
I hope you'll agree with support default gateways (with upper cost metric) over the LAN interface.

Regards.

Ysurac, 2019-05-01:
Not I'm not agree. Your configuration is too strange, nothing that I can do on OpenMPTCProuter. It's a router, only a router, you can set as you want manually but nothing that I will do.

lars18th, 2019-05-01:
OK,

Only two points to comment:

1. I can't _manually_ configure it in this way as you _enforce_ to remove the default gateway.
2. In my opinion OpenMPTCProuter can run as an _optional_ router, the only requirement is just this one.

Summarizing my request: please, support the **option** of "[ ] Not remove default gateway". It's a simple bool to check in your scripts before remove the default gateway.

Ysurac, 2019-05-01:
The default gateway of what ?
Optional router doesn't exist in a network.

lars18th, 2019-05-01:
> The default gateway of what ?

The default gateway in the "routes" of the OpenMPTCProuter.
Please, do this test: configure it manually inserting in the configuration of the network a "default gateway" in the LAN section entry (with some metric greater than 0). Then configure the MPTCP part. What you see then? That _your_ default route in `route -n` is removed! This is the pitfall that I suggest to remove with an optional configuration parameter.

> Optional router doesn't exist in a network.

You sure?
- In _any_ TCP/IP router you can put more than one default gateways using different metrics.
- And, using protocols like VRRP you can configure transparent default gateways.
- And nothing limits to have an environment where the default gateway of your network (a device that isn't the OpenMPTCProuter) routes to the OpenMPTCProuter (in the same LAN) and then it routes to outside. In this case your _true_ router routes to OpenMMPTCProuter only when it runs and has the MPTCP tunnel running.

Why not then?

Ysurac, 2019-05-01:
Sure LAN route is removed. Not problem here. As a router it would be quite strange to route LAN to LAN. At least it's not what OpenMPTCProuter is made for.
You can put multiple gateway with different metrics on client. Even with VRRP or what you want there is no "optional" router.
I really don't understand  what you want and why you can't do it with OpenMPTCProuter. You seems to want to use it only sometimes, but OpenMPTCProuter can use a connection only as backup if needed. OpenMPTCProuter need to control routes and need to know where to send packets, you can think about it more like a modem if you want.

As it's opensource, you can submit patches if you think there is a better solution.

lars18th, 2019-05-02:
Hi @Ysurac ,

Thank you for discussing it further. And I'm glad you're open-minded.

- LAN to LAN is not a problem. Why? Because in fact you're not "routing" to the LAN. The only "tricky" issue is that your "next" router (hop) is sharing the IP range of your LAN. But you're not sending packets to LAN devices. I hope you can understand this relevant point.

> Sure LAN route is removed.

- And this is the problem. If you remove the manual configuration from the user, then how we can achieve the objetive? Remember: my route line is like `ip route add default via 192.168.1.1 metric 100 dev br-lan`. and that will never interfere with your configuration, because the metric is too big.

> I really don't understand what you want and why you can't do it with OpenMPTCProuter. You seems to want to use it only sometimes, but OpenMPTCProuter can use a connection only as backup if needed. 

I describe in a more simple way:
- I have a main Router (it's not an OMR).
- I have one OMR in the same LAN.
- My router is controlling if packets goes out using the OMR or not.

Then I can use, for example, the OMR only for some services, or some devices, or whatever I really need. That's _not all traffic goes over OMR_ and another router is controlling it.

> As it's opensource, you can submit patches if you think there is a better solution.

Here the problem:
- You know well the internals of OMR, and I don't.
- You're still developing it, so futher changes will be done.
- What I request, is a simple OMR configuration value called `user_default_gw`, that when it's true (by default is false) then **you don't remove the LAN route when you configure the OMR**. It's just that. Nothing more!

Or if you know another alternative to achieve it, then please explain it.
I'm desperate for a solution to this problem.
Regards.