Ysurac/openmptcprouter #372: Suggestion: DNS-over-TLS (TCP)

lars18th, 2019-02-20:
Hi @Ysurac ,

In order to bypass completly UDP, and use only TCP (with Multipath), I suggest to configure the "Unbound DNS client/server" with DNS-over-TLS.

Here the description about "DNS-over-TLS" by Cloudflare:
https://blog.cloudflare.com/dns-over-tls-for-openwrt/

And here you can read about how to enable it when using Unbound in OpenWRT:
https://github.com/openwrt/packages/tree/master/net/unbound/files#how-to-tls-over-dns

I hope you agree! :wink:

Ysurac, 2019-02-20:
This will not be added.
DNS-over-TLS is only supported by Google and Cloudflare public DNS, it's not really a so good idea for privacy.
This will not be the default for now.

closed 2019-02-20 by Ysurac

lars18th, 2019-02-21:
Hi @Ysurac ,

> This will not be added.

Sorry, why not?
- I'm not suggesting to make it by default, only to support it (as an option).
- DNS-over-TLS is by far (I do a lot of tests) more fast than DNS-over-UDP-over-TCP_TUNNEL. So using a persistent HTTPS connection to the DNS server over MPTCP we obtain fast DNS resolutions.
- The current "Unbound" package has already DNS-over-TLS support.
- A lot of users already use 8.8.8.8 and 1.1.1.1 as DNS servers, so for these users this isn't a problem.

Please, consider it!

Ysurac, 2019-02-21:
It's supported, you only need to configure unbound for that

dnwk, 2019-02-25:
@[lars18th](/lars18th) I am using DNS-over-TLS right now. I install Stubby and have ubound forward the query via Stubby.

lars18th, 2019-02-26:
> @[lars18th](/lars18th) I am using DNS-over-TLS right now. I install Stubby and have unbound forward the query via Stubby.

Hi @dnwk ,

Great! Please, explain your configuration in detail?
https://openwrt.org/docs/guide-user/services/dns/stubby

I hope @Ysurac will consider to include this "option" (as optional, that's only supported and not enforced).

dnwk, 2019-02-28:
@lars18th   I install stubby  in System->Software and then in Network->DNS  DNS forwardings to  127.0.0.1#5453

dnwk, 2019-02-28:
@lars18th  Stubby comes with default configuration to Cloudflare's 1.1.1.1   Unless you want to use a different provider, there is minimal configuration.  I think it's better use of @Ysurac's time not to develop anything if there is a way to preserve the Stubby when upgrading to new version.

lars18th, 2019-02-28:
> I think it's better use of @Ysurac's time not to reinvent the wheel if there is a way to preserve the Stubby when upgrading to new version.

I totally agree!

However, OMR uses some scripts+tools that already exists. So, to not reinvent the wheel and support it, perhaps it will be usefull to include `stubby` by default and let in the config of OMR the optioin to enable it or not (and off by default).

I hope @Ysurac will agree (even though I don't think he wants to). :thinking:

Ysurac, 2019-02-28:
Unbound should be enough to support DNS-over-TLS. I will test this.

dnwk, 2019-02-28:
@Ysurac I found a Unbound configuration for DNS-over-TLS. However, I don't know if there is a way to manage it over web interface or I need to modify the config file. And if config file I customized will survived an system version upgrade. 

https://www.dnsknowledge.com/unbound/configure-unbound-dns-over-tls-on-linux/
#Adding DNS-Over-TLS support
server:
forward-zone:
name: "."
forward-ssl-upstream: yes
## Cloudflare DNS
forward-addr: 1.1.1.1@853
forward-addr: 1.0.0.1@853
## Also add IBM IPv6 Quad9 over TLS
forward-addr: 9.9.9.9@853
forward-addr: 149.112.112.112@853
## IPv6  Cloudflare DNS over TLS
forward-addr: 2606:4700:4700::1111@853
forward-addr: 2606:4700:4700::1001@853

dnwk, 2019-02-28:
@lars18th   https://blog.cloudflare.com/dns-over-tls-for-openwrt/   Here are some details on how to configure it on OpenWRT. So, I guess SSH into OpenWRT and modify config is necessary.

lars18th, 2019-03-01:
Hi,

For me isn't a problem to *manually* configure it. :laughing: 
However, I feel other users will like to use it, as it will be more efficient (fast) as it's TCP native.
So, I suggest to:
* Re-open this issue (please, @Ysurac )
* Put some info about it (i.e. How to use DNS-Over-HTTP or DNS-Over-TLS) as a refence in the documentation.
* And only if you found it useful, then add _some_ support.

In any case, I recommend to use DoH as it's by far more easy to configure and more fast to work (it uses HTTP persistent connections, so it's very fast over MPTCP).

Regards.