Ysurac/openmptcprouter #373: Interface SQM/QoS: Troubles

lars18th, 2019-02-20:
Hi @Ysurac ,

In my first configuration I use some values for "Download speed" and "Upload speed" for WAN1 & WAN2. These values are indicated in Kb/s (kilobits per second, so 6000 are around 6Mbit/s). And I use _different_ values for each one (WAN1 has around half the speed of WAN2 in download).

However, when I set these values, the total throughput is very bad. After setting all to "0" the multipath-TCP goes at full speed (checked with "iperf3").

So, I recommend to not use these values, until you check them. In my configuration both connections will have the same performance... and correspond to the values of WAN1.

My check is totally _local_. WAN1 & WAN2 are PPPoE. So I disable the mpTCP (`sysctl -w net.mptcp.mptcp_enabled=0`), and I start one iperf3 server in _the same_ PPPoE server (so the connection is full LAN). Then, I execute the client in the OMR using just one connection (with `-B <addr>` parameter). Setting the SQM/QoS both connections are limited to a very low value, and both use the same ratios. I feel something is wrong in the configuration.

Please, can you check it?
Regards.

lars18th, 2019-02-21:
Hi @Ysurac ,

After some testing I see that:

- When configuring with the wizard, the correct values are applied. I can see it in the page "NETWORK/SQM QoS/Queues/Basic Settings". However, the "Interface name" is *INCORRECT*. You apply it to the _DEVICE_ and not to the _INTERFACE_. So in fact in your scripts you're using the "device" and not the "interface". However, if I change it manually to the correct interface, then the correct values are applied.

I discovered this because my PPPoE share the same device (a bridge over a physical connection). So this is related to my problem #368 .

I hope you can fix it.
Regards.

lars18th, 2019-02-21:
Hi @Ysurac ,

I see here that you're getting from uci the `ifname` of the interface:
https://github.com/Ysurac/openmptcprouter-feeds/blob/877687892f9043956830ec3a9a1aa5f4ae4bd450/luci-app-openmptcprouter/luasrc/controller/openmptcprouter.lua#L27

So, when the _DEVICE_ is shared for different _INTERFACES_ (for example, when the device it's a Bridge), then you can mix them.

I suggest to use *only* device names when it's required. And use the interface name when you can. For example, for setting the interface in `sqm.wan1.interface` and `sqm.wan2.interface`.

FYI, when you can use the command `ifstatus <name>` with the name of the interface, then the OpenWRT environment can use it as a valid _INTERFACE_ and it's not required to use the _DEVICE_ name. Example, `sqm.wan1.interface="pppoe-wan1"` is valid, so `sqm.wan1.interface="br-lan"` isn't required (when the "pppoe-wan" interface is running on top of "br-lan").

Please, try it!

lars18th, 2019-02-21:
Hi @Ysurac ,

The solution is to use `l3_device` instead of `device` when you use uci.

Please, try to update your code!

Ysurac, 2019-02-21:
You added your interface manually, so you have to update sqm manually, else it's default settings.

closed 2019-02-21 by Ysurac