Ysurac/openmptcprouter #374: Bypass: whitelist and force

lars18th, 2019-02-20:
- OpenMPTCProuter version: openmptcprouter v0.46.3
- OpenMPTCProuter VPS version: NA
- OpenMPTCProuter platform: x86_64

Hi @Ysurac ,

With the current implementation of the "Bypass" functionality only _Whitelists_ are supported. However, in my case I want to do the opposite. That's I like to "Bypass all _except_ the entries in the list".

So, for example I want to define "bypass network 0.0.0.0/0.0.0.0" and after "force network 192.168.0.0/16". Then all traffic by bypassed except for the target networks of 192.168.*.0.

You can implement this, please?

lars18th, 2019-02-20:
Hi,

Just another suggestion: Please, support wildcards in the "Bypass/Domain". I need to bypass "*.openwrt.org" but it doesn't work!

Regards.

Ysurac, 2019-02-20:
This will no be added.
If domain is added as "openwrt.org" all subdomains are bypassed too but OpenMPTCProuter must be used as DNS.

closed 2019-02-20 by Ysurac

lars18th, 2019-02-21:
> If domain is added as "openwrt.org" all subdomains are bypassed too but OpenMPTCProuter must be used as DNS.

OK. Thank you for the tip! :+1: 

>> So, for example I want to define "bypass network 0.0.0.0/0.0.0.0" and after "force network 192.168.0.0/16". Then all traffic by bypassed except for the target networks of 192.168.*.0.

> This will no be added.

I accept it. However, in order to manually implement it (or something similar), can you explain, please, where are the config files for this bypass functionality?