Ysurac/openmptcprouter #376: Tracker doesnt work if interface goes down

its27, 2019-02-22:
## Expected Behavior

I have two WAN connections, one is ethernet and the other wifi. If I disconnect the ADSL cable from the router providing the wifi connection the tracker works as expected.

Fri Feb 22 12:00:37 2019 user.notice post-tracking-post-tracking: wlan0 switched off because ping from 192.168.1.222 error (80.67.169.40,4.2.2.1)
Fri Feb 22 12:00:37 2019 user.notice post-tracking-post-tracking: Delete vps (68.183.43.173) route via 192.168.1.254 dev wlan0
Fri Feb 22 12:00:37 2019 user.notice post-tracking-post-tracking: wwan down. Replace default route by 192.168.43.141 dev wlan1

However if I power off the router providing the WiFi so the wireless radio in unassociated with the access point the tracker does not detect it down and make the necessary changes. 

Fri Feb 22 12:03:17 2019 daemon.notice netifd: Network device 'wlan0' link is down
Fri Feb 22 12:03:17 2019 daemon.notice netifd: Interface 'wwan' has link connectivity loss

## Specifications

  - OpenMPTCProuter version: 0.46.3
  - OpenMPTCProuter VPS version: 0.93
  - OpenMPTCProuter platform: RPI2

Ysurac, 2019-02-22:
I would need the result of ```uci show``` via SSH on OpenMPTCProuter (remove at least IP of the server).

its27, 2019-02-23:
Please find below.

snmpd.public_access.notify='none'
snmpd.private_access=access
snmpd.private_access.group='private'
snmpd.private_access.context='none'
snmpd.private_access.version='any'
snmpd.private_access.level='noauth'
snmpd.private_access.prefix='exact'
snmpd.private_access.read='all'
snmpd.private_access.write='all'
snmpd.private_access.notify='all'
snmpd.@system[0]=system
snmpd.@system[0].sysLocation='office'
snmpd.@system[0].sysContact='bofh@example.com'
snmpd.@system[0].sysName='OpenMPTCProuter'
snmpd.@exec[0]=exec
snmpd.@exec[0].name='filedescriptors'
snmpd.@exec[0].prog='/bin/cat'
snmpd.@exec[0].args='/proc/sys/fs/file-nr'
snmpd.@engineid[0]=engineid
snmpd.@engineid[0].engineidtype='3'
snmpd.@engineid[0].engineidnic='eth0'
snmpd.general=snmpd
snmpd.general.network='lan'
snmpd.general.enabled='0'
snmpd.general.ipv6cpipv4='1'
sqm.wan1=queue
sqm.wan1.qdisc='fq_codel'
sqm.wan1.script='simple.qos'
sqm.wan1.qdisc_advanced='0'
sqm.wan1.linklayer='none'
sqm.wan1.enabled='0'
sqm.wan1.download='0'
sqm.wan1.upload='0'
sqm.wan1.debug_logging='0'
sqm.wan1.verbosity='5'
sqm.wan1.interface='eth0'
sqm.wan2=queue
sqm.wan2.qdisc='fq_codel'
sqm.wan2.script='simple.qos'
sqm.wan2.qdisc_advanced='0'
sqm.wan2.linklayer='none'
sqm.wan2.enabled='0'
sqm.wan2.download='0'
sqm.wan2.upload='0'
sqm.wan2.debug_logging='0'
sqm.wan2.verbosity='5'
sqm.wan2.interface='eth0'
sqm.wwan=queue
sqm.wwan.qdisc='fq_codel'
sqm.wwan.script='simple.qos'
sqm.wwan.qdisc_advanced='0'
sqm.wwan.linklayer='none'
sqm.wwan.debug_logging='0'
sqm.wwan.verbosity='5'
sqm.wwan.enabled='1'
sqm.wwan2=queue
sqm.wwan2.qdisc='fq_codel'
sqm.wwan2.script='simple.qos'
sqm.wwan2.qdisc_advanced='0'
sqm.wwan2.linklayer='none'
sqm.wwan2.debug_logging='0'
sqm.wwan2.verbosity='5'
sqm.wwan2.enabled='1'
system.@system[0]=system
system.@system[0].hostname='OpenMPTCProuter'
system.@system[0].timezone='UTC'
system.@system[0].log_size='64'
system.@system[0].urandom_seed='0'
[... 169 lines cut ...]
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.hwmode='11g'
wireless.radio1.path='platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.0'
wireless.radio1.htmode='HT20'
wireless.radio1.disabled='0'
wireless.radio1.country='00'
wireless.radio1.legacy_rates='1'
wireless.radio1.channel='2'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].network='wwan2'
wireless.@wifi-iface[0].ssid='OnePlus 5'
wireless.@wifi-iface[0].encryption='psk2'
wireless.@wifi-iface[0].device='radio1'
wireless.@wifi-iface[0].mode='sta'
wireless.@wifi-iface[0].bssid='94:65:2D:83:E9:22'
wireless.@wifi-iface[0].key=''
wireless.@wifi-iface[1]=wifi-iface
wireless.@wifi-iface[1].network='wwan'
wireless.@wifi-iface[1].ssid='BTHub6-xxx'
wireless.@wifi-iface[1].encryption='psk2'
wireless.@wifi-iface[1].device='radio0'
wireless.@wifi-iface[1].mode='sta'
wireless.@wifi-iface[1].bssid='BC:30:D9:6A:56:11'
wireless.@wifi-iface[1].key=''

Ysurac, 2019-02-23:
It's not all the data...

its27, 2019-02-23:
Sorry

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].network='lan'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@zone[1].network='wan wan6 wan1 wan2 wwan2 wwan'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
[... 954 lines cut ...]
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.hwmode='11g'
wireless.radio1.path='platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.0'
wireless.radio1.htmode='HT20'
wireless.radio1.disabled='0'
wireless.radio1.country='00'
wireless.radio1.legacy_rates='1'
wireless.radio1.channel='2'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].network='wwan2'
wireless.@wifi-iface[0].ssid='OnePlus 5'
wireless.@wifi-iface[0].encryption='psk2'
wireless.@wifi-iface[0].device='radio1'
wireless.@wifi-iface[0].mode='sta'
wireless.@wifi-iface[0].bssid='94:65:2D:83:E9:22'
wireless.@wifi-iface[0].key=''
wireless.@wifi-iface[1]=wifi-iface
wireless.@wifi-iface[1].network='wwan'
wireless.@wifi-iface[1].ssid='BTHub6-2C9R'
wireless.@wifi-iface[1].encryption='psk2'
wireless.@wifi-iface[1].device='radio0'
wireless.@wifi-iface[1].mode='sta'
wireless.@wifi-iface[1].bssid='BC:30:D9:6A:56:11'
wireless.@wifi-iface[1].key=''

Ysurac, 2019-02-23:
Can you give me the result of ```ifstatus wwan``` via SSH on OpenMPTCProuter ?

its27, 2019-02-28:
With wireless connected

{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 80,
        "l3_device": "wlan0",
        "proto": "dhcp",
        "device": "wlan0",
        "updated": [
                "addresses",
                "routes",
                "data"
        ],
        "metric": 6,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "192.168.1.222",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [
                {
                        "target": "0.0.0.0",
                        "mask": 0,
                        "nexthop": "192.168.1.254",
                        "source": "192.168.1.222\/32"
                }
        ],
        "dns-server": [
                "192.168.1.254"
        ],
        "dns-search": [
                "home"
        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ]
        },
        "data": {
                "hostname": "OpenMPTCProuter",
                "leasetime": 86400

With wireless router turned off (not associated with AP)

  "up": false,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "dhcp",
        "device": "wlan0",
        "data": {

closed 2019-07-04 by Ysurac