Ysurac/openmptcprouter #377: Change OMR JSON port 65500

kapook007, 2019-02-24:
how to change OpenMPTCProuter OMR JSON port 65500  ?

for multi OpenMPTCProuter  with 1ip nat multi vps.

OpenMPTCProuter 1     >                                      > vps1
OpenMPTCProuter 2     >         IDC 1IP  with nat     > vps2
OpenMPTCProuter 3     >                                      > vps3

Ysurac, 2019-02-24:
Multi OpenMPTCProuter on one VPS is not supported for now.

closed 2019-02-24 by Ysurac

kapook007, 2019-02-24:
not Multi OpenMPTCProuter with 1vps
i mean 1:1
i have 3 OpenMPTCProuter and 3 vps with 1 ip idc

Ysurac, 2019-02-24:
ok, not sure it's the best idea and it's also not supported.
Else on OpenMPTCProuter VPS port is defined with openmptcprouter.vps.port, but you will need to change also shadowsocks and glorytun ports on the VPS, but they are fixed in omr-admin python script.

kapook007, 2019-02-24:
i can change port shadowsocks and glorytun  in webconfig .
 OMR JSON port 65500 , where are omr-admin python script in  OpenMPTCProuter?

Ysurac, 2019-02-24:
omr-admin is on the VPS, and ports are set to OpenMPTCProuter by this script. It's in /usr/local/bin/

kapook007, 2019-02-24:
ok thank you ^^

kapook007, 2019-02-24:
i got in vps  /usr/local/bin/  omr-admin
but  OpenMPTCProuter no have /usr/local/bin/ .
how to change OMR JSON port in OpenMPTCProuter ?

Ysurac, 2019-02-24:
As I said, on OpenMPTCProuter, the config is in uci openmptcprouter.vps.port, so a ```uci set openmptcprouter.vps.port=24564``` should work.