Ysurac/openmptcprouter #379: Is some OpenMPTCPRouter feature relies on OpenMPTCPRouter.com?

dnwk, 2019-02-25:
While the website of OpenMPTCPRouter.com went down on Sunday, the router won't be able to get public IP and post-tracking-post-tracking:Get status and settings for wan 1 show up failed in system log. 

Is there a way to make calling out to external website optional or configurable? I hope the function of the router should not be depend on an external website.

Ysurac, 2019-02-25:
ip.openmptcprouter.com is used to detect server public IP (no really other choice to get public IP).
But when the server was done this shouldn't impact any router because other sites should be used in this cas. I will check where the problem is.
I will also make this fully configurable ;)

dnwk, 2019-02-25:
post-tracking-post-tracking:Get status for wan 1 fail shows up in the log. Is it also related to openmptcprouter.com went down?   Because those warning in the log goes away once site is back up.

Ysurac, 2019-02-26:
Yes it's related. This script also get IP of each wan.

Randdrick, 2019-02-27:
Hello,
I think there is also a problem with admin server scripts. (see capture image)
[image]

flimo44, 2019-02-28:
Hello,
I have same problem with admin server script.

Ysurac, 2019-02-28:
It's not related.

user747, 2019-03-04:
Does it still call if omr-tracker disabled? 

https://github.com/Ysurac/openmptcprouter-feeds/blob/09b2d20c2f46ac03a2aa861e4df6d84bb2f2b5bf/luci-app-openmptcprouter/luasrc/controller/openmptcprouter.lua#L702

Ysurac, 2019-03-04:
Yes in status page.
You have to wait for next version to have a checkbox to disable IP check.

user747, 2019-03-04:
> Yes in status page.
> You have to wait for next version to have a checkbox to disable IP check.

Could also have option to switch server default too. 

`wget -4 -qO- -T 2 http://checkip.dyndns.org/`

Ysurac, 2019-03-13:
Checkbox available in OpenMPTCProuter v0.47

closed 2019-03-13 by Ysurac

user747, 2019-04-12:
On v0.48 rpi3 squashfs

Check
['Disable external check'](https://github.com/Ysurac/openmptcprouter-feeds/blob/cf89a46736334d6aa1957d6fdab2ac36bd35e5ef/luci-app-openmptcprouter/luasrc/view/openmptcprouter/settings.htm#L134) 

Save & Apply
Page refreshes
No change

root@OpenMPTCProuter:/tmp# uci -q get openmptcprouter.settings.external_check
1

I think after change, it should be "0" ? 

Anything but 0 and it runs like default
if uci:get("openmptcprouter","settings","external_check") ~= "0" then

Ysurac, 2019-04-12:
This will be fixed in next release.

In omr-tracker this doesn't call the website, only check if it can be resolved (and only in the case user want dns check).

Ysurac, 2019-04-12:
Option to choose between website to check the IP is not so easy: my website return only the IP, for dyndns there is some text, so I would need to parse that...

dnwk, 2019-04-13:
Try this http://myip.cpanel.net/

user747, 2019-04-15:
Another one 
`curl ifconfig.co`